carla-simulator/carla #2748: Independent steering for front and rear axle

lisatenbo, 2020-04-15:
Hello,
I want to make a vehicle **follow a specific path** which I defined beforehand. It should be possible, to **steer the rear axle** independently of the front steering. Can I fulfill this task with CARLA, and if yes, what would be the best way? 
Is it also possible if I do not use Unreal Engine for this?
Thank you in advance!

germanros1987, 2020-06-26:
@lisatenbo you can't as the platform is right now. Do you really need to control the front and rear axles independently? What is your use case?

germanros1987, 2020-07-30:
@lisatenbo are you still interested in this issue?

lisatenbo, 2020-07-31:
I'm not interested anymore.

closed 2020-07-31 by germanros1987

dodosch, 2021-04-15:
Hi @germanros1987 , has this feature been implemented yet?

CharlesXu1124, 2022-03-19:
@germanros1987 I am also interested, please feel free to tell us a way to control them separately!

Nisarg236, 2022-07-21:
> 

I am trying to simulate a robot in carla that works on swerve drive. So is there any feature to control steering angles of each wheel through ROS bridge or python API. If not then how can I add it by changing the source code? can anyone help me on this please

edmc25, 2025-04-25:
I'm also interested in the rear axle steering. Is it maybe already implemented?