carla-simulator/carla #2752: docstrings for carla

HarryWindsor, 2020-04-16:
Is is possible to include the docs from https://carla.readthedocs.io/en/latest/python_api/#carla.Actor in the python library? I would prefer using the docs in my IDE rather than jumping into the browser when I have to look something up.

Thank you.

pperle, 2020-04-20:
You could try using [Pycharms external documentation](https://www.jetbrains.com/help/pycharm/settings-tools-python-external-documentation.html#) feature. The URL should look something like that `https://carla.readthedocs.io/en/latest/python_api/#carla.{class.name}.{function.name}`.

sergi-e, 2020-04-23:
Hi, @HarryWindsor !

Did the solution provided by @pperle served you well?

Alby407, 2020-04-24:
This does not work. PyCharm won't give any information when for example hovering over `carla.World`. If I manually go to e.g. `https://carla.readthedocs.io/en/latest/python_api/#carla.World`, it works.

pperle, 2020-05-01:
@Alby407 you are right, the URL I posted does not work for classes, only for their functions. For classes, there is one dot to many `https://carla.readthedocs.io/en/latest/python_api/#carla.World.`.
I think the right parameter to use would be `{element.qname} - full-qualified name of element under caret`. The problem is that the fully-qualified name is not `carla.World` but `carla.libcarla.World` which does not match the doc URL (and even this would not fix the problem, looks like PyCharm has problems parsing the docs).

The right solution would be to include the docs.

pperle, 2020-05-01:
Inspired by [this post](https://stackoverflow.com/a/34898716/9309705) I tried to add the docstring from `carla/PythonAPI/docs/*.yml` to the python stub. You can find the code here: https://github.com/pperle/carla/tree/docstrings_py . It is hacky, I know but it works well enough for me at the moment.

If anyone is interested here is the compiled Linux version [carla-0.9.9-py3.5-linux-x86_64.egg.zip](https://github.com/carla-simulator/carla/files/4565282/carla-0.9.9-py3.5-linux-x86_64.egg.zip). The only thing that does not work are the type hints.

Example preview in PyCharm:
[image]

germanros1987, 2020-06-26:
@pperle you have ran 90% of this marathon...would you like to finish it by making a PR to CARLA?

closed 2020-09-05 by stale[bot]

BruceXSK, 2021-11-24:
I just found that you can extract the .egg file to the site-packages dir of your python. And PyCharm can generate some simple python stubs from the .so library. The stub files generated are very naive that only the names of classes and methods are included, but it is enough for coding.