carla-simulator/carla #2755: Advanced highway Ai and Road System in CARLA using Unreal Engine

ujjawal-kumar, 2020-04-16:
Is it possible to create a new map(Advanced highway Ai and Road System) in CARLA 
using an Unreal Engine without using Roadrunner by VectorZero?

I found one youtube video about it.
[https://youtu.be/6g2IgGxnUrg?list=PLjRSTRCPlfGDafjEydOggrck9W6VBoovr](https://youtu.be/6g2IgGxnUrg?list=PLjRSTRCPlfGDafjEydOggrck9W6VBoovr)

XGodina, 2020-04-29:
Hi @ujjawal-kumar.
Well, we are using the RoadRunner, because it provides us the OpenDrive Information and .fbx files. However, in our last release 0.9.9, we can generate road through OpenDrive files.
https://www.youtube.com/watch?v=dvPhN0Gbb0w

mikemwang, 2020-04-30:
So is the answer to @ujjawal-kumar 's question **yes**? Roadrunner is no longer available to the general public after being taken over by MathWorks (I've requested a trial from them, no response yet), is the solution to use Unreal Engine?

XGodina, 2020-05-03:
@mikemwang If you want to create a HIghway with Unreal, there are a lot of tutorials to do it. 
I don't know if you need the OpenDrive information, in that case, you need software to represent this information.

germanros1987, 2020-06-26:
@mikemwang @ujjawal-kumar I am going to unpack all those questions:

- [Q] Can I use tool X to create a map?
- [A] Sure, why not... But we probably don't provide support for it yet. So I would like to invite you to build a bridge to CARLA and contribute it back to the community.
- [Q] Well, then what is the right way to create a map directly in UE4?
- [A] By making sure that an OpenDRIVE file is generated. If you don't use the tools that we provide to do this, then it is your responsibility to find the right way to generate the OpenDRIVE file. This is the core element to enable navigation.
- [Q] RoadRunner was acquired by Mathworks, are we all doomed and won't be able to use RoadRunner ever again?
- [A] No! If you are a student/researcher you will get your license. Sometimes they may take some time...people get busy... If you keep having problems send us an e-mail at info-carla@osvf.org and we will happily help you with that.

closed 2020-09-05 by stale[bot]