carla-simulator/carla #2757: bad_alloc in CppClient example

cheshirekow, 2020-04-16:
I installed carla via the debian package from http://dist.carla.org/carla-0.9.8/ and I built the C++ client library from source on commit 3a36b1637804bccc7b1dc82ef3dce607fd90ba14. When I try to run the C++ example from `Examples/CppClient` (built with `make` in that directory) the program crashes with:
```
Client API version : 0.9.8-269-g3a36b163
Server API version : 0.9.8
Loading world: /Game/Carla/Maps/Town02
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.8-269-g3a36b163 
WARNING: Simulator API version  = 0.9.8 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted
```

I built the client in debug mode and executed under GDB to get a stacktrace which shows the following:

```
[Switching to Thread 0x7ffff2fd1700 (LWP 11253)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff6c95801 in __GI_abort () at abort.c:79
#2  0x00007ffff7688957 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff768eae6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff768eb21 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff768eda9 in __cxa_rethrow () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00005555558199ce in std::_Hashtable<unsigned int, std::pair<unsigned int const, carla::client::ActorSnapshot>, std::allocator<std::pair<unsigned int const, carla::client::ActorSnapshot> >, std::__detail::_Select1st, std::equal_to<unsigned int>, std::hash<unsigned int>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true> >::_M_rehash(unsigned long, unsigned long const&) ()
#7  0x00005555558197f0 in carla::client::detail::EpisodeState::EpisodeState(carla::sensor::data::RawEpisodeState const&) ()
#8  0x000055555581413c in std::_Function_handler<void (carla::Buffer), carla::client::detail::Episode::Listen()::{lambda(auto:1)#1}>::_M_invoke(std::_Any_data const&, carla::Buffer&&) ()
#9  0x000055555583fd29 in boost::asio::detail::completion_handler<carla::streaming::detail::tcp::Client::ReadData()::{lambda()#1}::operator()() const::{lambda(boost::system::error_code, unsigned long)#1}::operator()(boost::system, unsigned long) const::{lambda()#1}>::do_complete(void*, boost::asio::detail::scheduler_operation*, boost::system const&, unsigned long) ()
#10 0x00005555557f6ff8 in boost::asio::detail::scheduler::run(boost::system::error_code&) ()
#11 0x00005555557f90e0 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<carla::ThreadPool::AsyncRun(unsigned long)::{lambda()#1}> > >::_M_run() ()
#12 0x00007ffff76b96df in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#13 0x00007ffff7bbd6db in start_thread (arg=0x7ffff2fd1700) at pthread_create.c:463
#14 0x00007ffff6d7688f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
Is this a known issue? Is there a known work-around?

nsubiron, 2020-04-17:
Client and server versions need to match, otherwise you get this sort of errors. Client and server share a lot of raw data, it's not safe to ignore the warning unless you know for sure none of the networking code changed
```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
```

cheshirekow, 2020-04-17:
I guess I'm a little confused. Why is there a version mismatch between `0.9.8` and `0.9.8`?

nsubiron, 2020-04-17:
It is between `0.9.8` and `0.9.8-269-g3a36b163`. These tags are extracted from git when you build, so `0.9.8` means the server was built at the commit pointed by that git tag, and `0.9.8-269-g3a36b163` means you built the client at commit `3a36b163`, 269 commits ahead of 0.9.8. So potentially a lot of changes in between.

gitmaster3000, 2020-04-20:
I was having the same issue but couldn't resolve it. The problem is when I build from source the client version never goes beyond 0.9.8

nsubiron, 2020-04-20:
@gitmaster3000 This is probably because you have the "carla" Python package installed, so when you run the scripts this version is imported instead of the one you just built. Try

    python -m pip uninstall carla

(using the same Python version you use with Carla).

gitmaster3000, 2020-04-20:
OH MY GOD! You sir are jesus christ reincarnated.. THREE WEEKS, literally three weeks I have struggled with this and now it's solved. daaamn

cheshirekow, 2020-04-20:
> The mismatch appears between `0.9.8` and `0.9.8-269-g3a36b163`. These tags are extracted from git when you build, so `0.9.8` means the server was built at the commit pointed by that git tag, and `0.9.8-269-g3a36b163` means you built the client at commit `3a36b163`, 269 commits ahead of 0.9.8. So potentially a lot of changes in between.

Oh I see. I was operating under the assumption that the extra bits were due to building from source and if there was an API change than one of the first three numbers would have changed. I understand now. 

So I built the client from tag `0.9.8` and no longer see this `bad_alloc`, but I do see a segmentation fault when trying to use the waypoint API:

```
Thread 1 "carla-planner" received signal SIGSEGV, Segmentation fault.
0x0000555555815e16 in carla::client::Waypoint::GetNextUntilLaneEnd(double) const ()
(gdb) bt
```
Shall I describe this further here or shall I open a new issue?

nsubiron, 2020-04-21:
@cheshirekow yeah, that looks like something else, better open a new issue.

closed 2020-04-23 by sergi-e