carla-simulator/carla #2763: Could we use mouse to interact with carla?

ZhengZhongyuan11, 2020-04-18:
I just want to know is there any way that we can enable mouse in carla and use the mouse to interact with the carla world? Just like click on a point in the map and get the location information of that point?

sergi-e, 2020-04-23:
Hi, @ZhengZhongyuan11 !

Unfortunately, this kind of feature is not implemented in CARLA. 
However, should I guess that you are working with a CARLA package? If so, maybe you could consider moving to a build from source. That would allow you to open CARLA with the Unreal Editor, and maybe provide you with the information you need, depending on the exact intentions. 
Here is a link to the docs, where the build process for Linux and Windows are explained: https://carla.readthedocs.io/en/latest/

ZhengZhongyuan11, 2020-05-01:
> Hi, @ZhengZhongyuan11 !
>
> Unfortunately, this kind of feature is not implemented in CARLA.
> However, should I guess that you are working with a CARLA package? If so, maybe you could consider moving to [...]

ZhengZhongyuan11, 2020-05-01:
> Hi, @ZhengZhongyuan11 !
>
> Unfortunately, this kind of feature is not implemented in CARLA.
> However, should I guess that you are working with a CARLA package? If so, maybe you could consider moving to [...]

OK I will try it, thanks!

closed 2020-05-01 by ZhengZhongyuan11