carla-simulator/carla #2766: Warning: cannot parse georeference: ''

praetorianer777, 2020-04-20:
Hello,
how can I get rid of this warning message? I am using Carla 0.9.8.

sergi-e, 2020-04-23:
Hi, @praetorianer777 !

Could you provide with more information? When exactly does the message appear?

praetorianer777, 2020-04-24:
Hello,

I get this message when I request the map from the world.

sergi-e, 2020-04-27:
May I ask for a screenshot showing the terminal and the CARLA simulation running?

praetorianer777, 2020-04-29:
The simulation is a custom map made with roadrunner, attached are the screenshots.

[image]
[image]

germanros1987, 2020-06-26:
@praetorianer777 are you still having problems with this?

praetorianer777, 2020-06-26:
Hello, no it seems fine now! Thanks

closed 2020-06-26 by praetorianer777

patelchintan3112, 2021-12-09:
@praetorianer777 I am still facing this problem on Carla 0.9.11. How did you solve this problem?

praetorianer777, 2021-12-09:
Hello, sorry this was long ago, I don't know how I did solve the issue :-/

JoseZF, 2022-08-15:
i got the same warning when trying to create the navigation file for a custom map and when trying to run generate_traffic.py (only pedestrians) in the same map.
The function to get a random location from navigation returns None and therefore it cannot create a spawn point for pedestrians

[image]

bapukokare, 2022-08-17:
NEWS  direction not mentioned in open drive in file header description of xodr file respective map file