carla-simulator/carla #2767: Buggy physics controls?

praetorianer777, 2020-04-20:
Hello,

if I get the current physic_controls from a vehicle and apply these back to the same vehicle, then it just starts to bug and jump around, is this normal?

sergi-e, 2020-04-23:
Hi, @praetorianer777!

I understand that you are using get_physics_control() and then apply_physics_control. May I ask what are your intentions? The physics controller is mostly static information about physical properties of the vehicle, thus, their are not supposed to be updated constantly.

praetorianer777, 2020-04-24:
Hello,

I just tried to change some data, for instance steering_curve and gear_switch_time

sergi-e, 2020-04-27:
@praetorianer777 
The physics of the vehicle are not intended to be changed on the fly. You should either turn the physics off and make the changes, or do those changes before even spawning the vehicle. 
Alternatively, if you are working in a build from source, you can change the properties of the vehicle in the UE Editor.

praetorianer777, 2020-04-29:
Okidoki, than this issue is closed... thank you!

closed 2020-04-29 by praetorianer777

iasonastsaousis, 2025-06-20:
What if we wanted to create a specific brake torque curve? Because currently carla allows setting only a maxBrakeTorque for each wheel, but my actual car has regen brake which changes its maximumBrakeTorque based on motor RPM..

I tried doing this from the client, calling apply_physics_control multiple times, but as mentioned it wont allow the car to move. Any thoughts?