carla-simulator/carla #2768: No pedestrian/vehicle collision at relatively slow speed

ShahriarSefati, 2020-04-20:
Hi,

At slow speeds (~10 km/h), vehicles tend to push the pedestrians away rather than colliding with them. This behavior persists until a certain point where the collision is detected (examples attached.)

In case of the Crossbike model, the problem seems to be even worse (no collision at ~15 km/h). Is there a way to fix this behavior?

Issue #13 is related but I assume a lot has changed since that issue was opened. 
Currently using Carla 0.9.8 on Ubuntu 18.04.

Thanks!

[image]

[image]

germanros1987, 2020-06-26:
This is critical and needs a quick fix @jackbart94 @Axel1092

bernatx, 2020-07-02:
The problem is that it feels unrealistic because the pedestrian has not the animation that moves apart from the vehicle. We need to fix that behaviour, first the pedestrians have so many colliding boxes and we need to remove some of them. Then, we will try to fine tune the pedestrian avoiding strategy to make it more realistic.

KasraMokhtari2112, 2020-08-13:
Hi, I am experiencing exactly the same issue #3183. Since I am doing RL on pedestrian-ego vehicle collision, this is absolutely important to get an accurate feedback from the collision sensor. I really appreciate it if you guys can work on this issue. Thanks a lot. 
@ShahriarSefati Did you find a way to work around this issue?

Alby407, 2020-10-16:
There was this merge last year: https://github.com/carla-simulator/carla/pull/1183/files
But is not really a "pedestrian collision sensor", because it will check within a radius, which is not necessarily the bounding box of the ego vehicle. You have to tune it manually.

> The problem is that it feels unrealistic because the pedestrian has not the animation that moves apart from the vehicle. We need to fix that behaviour, first the pedestrians have so many colliding [...]

It's still required, even if pedestrians get pushed away. The collision should be detected at least.

closed 2021-04-27 by stale[bot]

gaoyinfeng, 2021-06-23:
Is this bug fixed?

truncs, 2022-12-24:
@bernatx  Let's say that there is an animation for pedestrians and they have a good strategy for avoidance, would this still happen at low vehicle speeds if the vehicle manages to hit the pedestrians?
I feel that there should be a configurable option in the sensor that allows for a stricter definition of a collision.