carla-simulator/carla #2769: Carla PythonAPI building macOS Mojave

b1456, 2020-04-21:
Hello!

I am attempting to build **_ONLY_** the PythonAPI for Carla and Python 3.7 on macOS. I do not wish to run the actual Carla simulation on macOS. I have a windows PC that I can run the simulation on but, I want the Python library installed on my Mac. I have pulled the repository and have attempted to run `make PythonAPI` but, I get an error: `clang 8 is required, but it's not installed.` I believe that I need to install clang 8 via `sudo port install clang-8.0` but I do not want to mess up Xcode as I also code apps on my machine. Any help would be appreciated.

marcgpuig, 2020-04-28:
Hi @bwees,
You could try to remove the following to avoid the check, and try to compile again?
https://github.com/carla-simulator/carla/blob/67684c5b0d56df1fe9fc54f59b650975afe6b759/Util/BuildTools/Setup.sh#L7-L10

bwees, 2020-04-28:
Ok. Will this compile everything or just the PythonAPI?

bwees, 2020-04-28:
So it no longer complains about clang 8 (Thank you!) but, it is now complaining about installing boost.
`Building Boost.Build engine with toolset clang... 
Failed to build Boost.Build build engine
Consult 'bootstrap.log' for more details
make: *** [setup] Error 1`

I cannot find `bootstrap.log` in `/var/logs` or any of the boost directories (ran recursive grep).

bwees, 2020-04-28:
So that fixed the clang issue! Thank you! Now it is complaining about the rpclib and having too many errors. Here is a part of the log:

```FAILED: CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o 
/Applications/Xcode.app/Contents/Developer/usr/bin/g++  -DASIO_STANDALONE -DRPCLIB_ASIO=clmdep_asio -DRPCLIB_FMT=clmdep_fmt -DRPCLIB_MAC -DRPCLIB_MSGPACK=clmdep_msgpack -I/Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/include -isystem /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/dependencies/include -fPIC -std=c++14 -stdlib=libc++ -I/Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/include/c++/v1 -Wl,-L/Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/lib -DBOOST_NO_EXCEPTIONS -DASIO_NO_EXCEPTIONS -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -mmacosx-version-min=10.14   -Wall -pedantic -Weverything -Wno-c++98-compat -Wno-c++98-compat-pedantic -Wno-padded -Wno-missing-prototypes -Wno-undef -pthread -std=gnu++11 -MD -MT CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o -MF CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o.d -o CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o -c /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc
clang: warning: -Wl,-L/Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/lib: 'linker' input unused [-Wunused-command-line-argument]
In file included from /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
/Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/include/c++/v1/atomic:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_ATOMIC
        ^
In file included from /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
In file included from /Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/include/c++/v1/atomic:543:
/Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/include/c++/v1/__config:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_CONFIG
        ^
In file included from /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
In file included from /Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/include/c++/v1/atomic:544:
/Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/include/c++/v1/cstddef:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_CSTDDEF
        ^
In file included from /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
In file included from /Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/include/c++/v1/atomic:544:
In file included from /Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/include/c++/v1/cstddef:38:
/Users/kristiwees/Downloads/carla/Build/llvm-8.0-install/include/c++/v1/version:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_VERSIONH
        ^
In file included from /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /Users/kristiwees/Downloads/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc
```

Not sure if this is the issue but, I commented out `sed  -i s/"3.9.0"/"3.5.0"/g ${RPCLIB_BASENAME}-source/CMakeLists.txt` due to the error of -i not able to be used with stdin.

marcgpuig, 2020-04-28:
You need `llvm-8` anyway. You could use a virtual machine or a Docker to compile it without compromising your already configured machine. Are you sure you cannot use version 8 with the current one that you have?
I'm not a macOS user and I cannot test it, sorry about that.

marcgpuig, 2020-04-28:
Related to #2797?

bwees, 2020-04-28:
It is exactly like #2797. Will try to install `llvm-8` and set the env variables to the correct paths later today.

bwees, 2020-04-28:
So I installed clang-8 and llvm-8 and have set those for the compilers in `Setup.sh` I still get the errors in #2797

germanros1987, 2020-06-26:
@geohot do you have any advice?

hansbothe, 2020-07-09:
Setup.sh line 204 adding "-Wno-reserved-id-macro" option supresses the warnings shown.

The following error messages seem to be related to C++13 added functions in stdlib, which indicates that the generation of Build/llvm-install... grabs older stdlib.h on MAC without those

hansbothe, 2020-07-09:
> Related to #2797?

definitely the same, see my comment there.

closed 2020-09-29 by stale[bot]

Hither1, 2022-06-22:
I guess just installing clang-8.0 may be a simpler fix