carla-simulator/carla #2775: Can't apply steer and speed predicted values on the vehicle

rehamessameltagoury, 2020-04-21:
I am working on Nividia model 2016 ,but can't find a way to apply the predicted values in real time on carla simulator.Is there anything that i can read to figure this out TIA

sergi-e, 2020-04-23:
Hi, @rehamessameltagoury! 

You can' t directly apply a speed to a vehicle in CARLA. 
Vehicles in CARLA are controlled by, what we call a carla.VehicleControl (https://carla.readthedocs.io/en/latest/python_api/#carlavehiclecontrol). This allows to drive the vehicle using realistic commands, such as throttle or steering. 
Can you provide more specific information on what are your intentions?

rehamessameltagoury, 2020-05-01:
I just wanted to make the vehicle move in a certain speed predicted from a model .Thanks for your help .

germanros1987, 2020-06-26:
@rehamessameltagoury you would have to create a controller (e.g. a PID) that takes that speed as input and uses it to control the vehicle through the CARLA API, which accepts throttle and steering commands. For further information, please refer to the [documentation](https://carla.readthedocs.io/en/latest/python_api/#carlavehiclecontrol)

closed 2020-06-26 by germanros1987