carla-simulator/carla #2776: make launch failure

sooroosh, 2020-04-21:
Hi everyone, 
I am trying to get carla running on windows 10. I have followed the instructions here:
https://carla.readthedocs.io/en/latest/build_windows/
I have installed all the dependencies (git,cmake,make and python) and made sure they are added to the system path. I have installed VS Community 2017 with all requirements as well as Unreal Engine 4.24.3.
I have downloaded the assets from : 

> http://carla-assets.s3.amazonaws.com/20200416_5568e18.tar.gz

unzipped them and placed them in : 

> carla\Unreal\CarlaUE4\Content\Carla 

When I run "make lunch" it almost succeeds to finish the job, however in the end I get a very general error : 

>  -[BuildCarlaUE4]: Building Unreal Editor...
> The system cannot find the path specified.
> make: *** [CarlaUE4Editor] Error 1.

This is the full output generated in the command line after running it:

> D:\carla>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  12
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "D:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "D:\carla\Build\zlib-install" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "D:\carla\Build\" --zlib-install-dir "D:\carla\Build\\zlib-install"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "D:\carla\Build\libpng-1.2.37-install" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "D:\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "D:\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "D:\carla\Build\recast-cdce4e-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "D:\carla\Build\" --toolset msvc-14.1 --version 1.72.0 -j 12
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "D:\carla\Build\boost-1.72.0-install" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 file(s) copied.
        1 file(s) copied.
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in D:\carla\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

   This script provides the assets for CARLA Latest
   You can download the assets from here:

       http://carla-assets.s3.amazonaws.com/20200416_5568e18.tar.gz

   Unzip it in the "D:/carla/Unreal\CarlaUE4\Content\Carla" folder.
   If you want another version, search it in D:/carla/Util\ContentVersions.txt.

-[BuildLibCarla]: [Batch params]: --server --client
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18362.
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done
-- Generating done
-- Build files have been written to: D:/carla/Build/libcarla-visualstudio
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  carla_server.vcxproj -> D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server.lib
  carla_server_debug.vcxproj -> D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server_debug.lib
  -- Install configuration: "Release"
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18362.
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done
-- Generating done
-- Build files have been written to: D:/carla/Build/libcarla-visualstudio
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  carla_client.vcxproj -> D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\Release\carla_client.lib
  carla_client_debug.vcxproj -> D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\Release\carla_client_debug.lib
  -- Install configuration: "Release"
-[BuildLibCarla]: LibCarla server has been successfully installed in "D:/carla/Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "D:/carla/PythonAPI\carla\dependencies"!
-[BuildCarlaUE4]: [Batch params]: --build
--build
ECHO is off.
-[BuildCarlaUE4]: Building Unreal Editor...
The system cannot find the path specified.
make: *** [CarlaUE4Editor] Error 1

I have spent almost a day figuring out what is going wrong so I would be happy if you could help me solve this.

hallettc, 2020-04-23:
I am having the same issue. 

When I go to build the newest version of the build on Windows (`make launch` in the x64 Native Tools Command Prompt for VS 2017) I get the following error: 

```
-[BuildLibCarla]: LibCarla server has been successfully installed in "E:/carla/Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "E:/carla/PythonAPI\carla\dependencies"!
-[BuildCarlaUE4]: [Batch params]: --build
--build
ECHO is off.
-[BuildCarlaUE4]: Building Unreal Editor...
The system cannot find the path specified.
make: *** [Util/BuildTools/Windows.mk:20: CarlaUE4Editor] Error 1
```

This is following all instructions on the Windows Build page, with all path variables set (cmake, make, python 3.7.6, and git) and all of the newest pulls from GitHub. All of the assets have been unzipped and extracted into the proper folders. This issue has persisted through 3 reinstallations over the past two days (including Unreal Editor, which is currently version 4.24.3). 

When I look at the line it is referencing in the Windows.mk file, this is what I see (ln 14-31):

```
# use PHONY to force next line as command and avoid conflict with folders of the same name
.PHONY: import
import: server
              @"${CARLA_BUILD_TOOLS_FOLDER}/Import.py" $(ARGS)

CarlaUE4Editor: LibCarla
              @"${CARLA_BUILD_TOOLS_FOLDER}/BuildCarlaUE4.bat" --build

launch: CarlaUE4Editor
              @"${CARLA_BUILD_TOOLS_FOLDER}/BuildCarlaUE4.bat" --launch

launch-only:
              @"${CARLA_BUILD_TOOLS_FOLDER}/BuildCarlaUE4.bat" --launch

package: PythonAPI
              @"${CARLA_BUILD_TOOLS_FOLDER}/Package.bat" --ue-version 4.24 $(ARGS)

.PHONY: docs
```

I am not sure how to fix the issue, and have tried all of the solutions that I could think of. I have confirmed that BuildCarlaUE4.bat does exist in the E:\carla\Util\BuildTools folder. I have tried to force it to run, but nothing happens when running the .bat file from the terminal window or the GUI.  I am running make launch from the Carla folder.

bernatx, 2020-04-24:
Hi,
Define the environment variable
`UE4_ROOT = <...your path...>\Epic Games\UE_4.24>`
In windows it should not be needed, but by some reason it seems that is not finding the UE4 path source from the registry.
We will check further the problem to fix it, or document well this way.

hallettc, 2020-04-24:
This absolutely fixes the problem! Thank you @bernatx !

sooroosh, 2020-04-24:
Solves my problem too, thanks !

closed 2020-04-24 by sooroosh

EugenM121, 2020-05-02:
@bernatx @hallettc , you mean evironment path like "C:\Program Files\Epic Games\UE_4.24". I have  the same problem, but i can not solve it by adding environment path

AChildFromBUAA, 2020-05-03:
@EugenM121 I added this path in user variables and system variables as "UE4_ROOT" and also added it in system PATH. I'm not sure which one makes this available. If you reopen your cmd and run "make launch" under carla root folder, it should be fine.

bernatx, 2020-05-04:
The variable **UE4_ROOT** can be a user variable if only you need it (no other users in your OS), or can be a system variable if all yours users need it. You don't need to put the path in the system **path** variable. 

The important thing is to restart any command line console you have open before setting the variable, otherwise those consoles will have the old variables, not the new one.

SriramKichugari, 2020-08-19:
> @EugenM121 I added this path in user variables and system variables as "UE4_ROOT" and also added it in system PATH. I'm not sure which one makes this available. If you reopen your cmd and run "make launch" under carla root folder, it should be fine.

Maybe your Unreal version is a different one.  So better check the path while installing and add it to the user variable.

abood002, 2021-02-07:
Where can I get the "UE_4.24" from?

Axel1092, 2021-03-03:
@abood002 check our documentation in https://carla.readthedocs.io/en/latest/ in the Linux/Windows Build guides to ge Unreal engine

RobTranquillo, 2021-12-03:
I have the same problem. But I set the `UE4_ROOT ` variable before compiling, as it says in the documentation. The difference with me, I think, is that the others here install their Unreal version whereas I have compiled the Carla Unreal version myself. Also as it says in the [documentation](https://carla.readthedocs.io/en/latest/build_windows/).

I think the problem arises here in [line 130 of BuildCarlaUE4.bat](https://github.com/carla-simulator/carla/blob/8854804f4d7748e14d937ec763a2912823a7e5f5/Util/BuildTools/BuildCarlaUE4.bat#L130)
Because an added output of `echo "%ROOT_PATH%Unreal/CarlaUE4/CarlaUE4.uproject"` shows simply `"Unreal/CarlaUE4/CarlaUE4.uproject"`. The variable `%ROOT_PATH%` seems to be empty.

ohashi3399, 2022-06-21:
@bernatx @Axel1092
I read the above threads. but I have same status as @abood002
I couldn't find any files named "UE_4.24". I want to clarify where my mistake was made. It would be very helpful if you could tell me which files I should make pass, not directory. Sorry for asking in a somewhat strange way. Thanks.

snehanjalikalamkar, 2022-06-22:
I had the same error as @sooroosh and @hallettc 
How I solved it is just by setting the environment variables as required in the BuildCarlaUE4.bat file, refer to @hallettc's comment. I had to add two path variables as follows:

`UE4_ROOT` - `D:\Unreal_CARLA_Patch` 
`ROOT_PATH` - `D:/CARLARepo/carla/`

BuildCarlaUE4.bat uses the following code
`call "%UE4_ROOT%**\**Engine\Build\BatchFiles\Build.bat"^
        CarlaUE4Editor^
        Win64^
        Development^
        -WaitMutex^
        -FromMsBuild^
        **"%ROOT_PATH%Unreal/CarlaUE4/CarlaUE4.uproject"**
    if errorlevel 1 goto bad_exit

    call "%UE4_ROOT%\Engine\Build\BatchFiles\Build.bat"^
        CarlaUE4^
        Win64^
        Development^
        -WaitMutex^
        -FromMsBuild^
        "%ROOT_PATH%Unreal/CarlaUE4/CarlaUE4.uproject"
    if errorlevel 1 goto bad_exit
`
Please mind the slashes as the reference to UE4_ROOT already makes uses of a **\**, but ROOT_PATH does not.

ohashi3399, 2022-06-22:
I solved the problem by properly setting the user environment path and setting options when build UE4.sln. Thanks.

FBanani, 2022-07-03:
hi 
I have some issue after making launch
1: when carlaUE4 initialize 75%  it shows a message with some errors about failed to find. 
[image]

then I press ok. the carlaUE4_unreal editor open but it shows message logs with many failed to load.
[image]

 I close it and press play but it shows " One or more blueprints has an unresolved compiler error, are you sure you want to Play in Editor?"
[image]

 I press play in editor and after a time the CarlaUE4 is closed with the box "the UE4_carlaUE4 editor has crashed" fatal error .
[image]

 what should I do?

ohashi3399, 2022-07-04:
Hi @Fatemeh san.
**Note that my OS is Windows10. If your OS are others, It may not useful.**
I didn't face to your error, but I hope my opinion will make your help.
First error that you shown arises from files of /Game/Carla/Static... . I guess this file must locate the specific path. I see correct path from build manual as below. https://carla.readthedocs.io/en/latest/build_windows/

> The assets will be downloaded and extracted to the appropriate location if have 7zip installed. If you do not have this software installed, you will need to manually extract the file contents to Unreal\CarlaUE4\Content\Carla.

I think following errors also arise from first error, please let you check the path first.
If you already locate correct path, It may be a kind of version imcompatible issue of BP(Blue Print) files that you already installed. There is correct version corresponding each carla version. You may need to reinstall "Content" file (including BP files). You can check the version from ```ContentVersions.txt``` also shown in the above link.

> From the root CARLA directory, navigate to \Util\ContentVersions.txt. This document contains the links to the assets for all CARLA releases.
Extract the assets in Unreal\CarlaUE4\Content\Carla. If the path doesn't exist, create it.
Extract the file with a command similar to the following:

Thanks.

ohashi3399, 2022-07-04:
I also leave the comment of possible way to solve.
I encountered last ```fatal error!``` window when I built Carla0.9.11.  There was no information about this in issues (and no error message, terrified), I avoided to solve the issue directly, I switched from Carla0.9.11 to Carla0.9.13(latest) and it worked. If you have no specific reasons for using current version, I recommend to switch the Carla version to the latest.

FBanani, 2022-07-05:
> 

hank you it is solved

Ezharjan, 2022-10-29:
> 

The root of UE4_ROOT varible set in environment as a varible is `/SOME_PLACE/CarlaUnreal`. Make sure that there is an `Engine` folder inside `CarlaUnreal`.

Ehab-Iqnaibi, 2024-10-28:
I'm trying to build Carla Latest Release([CARLA 0.9.15], git clone https://github.com/carla-simulator/carla) on windows 11, with UE 4.26.2

make launch
ECHO is off.
ERROR: The system was unable to find the specified registry key or value.

-[BuildCarlaUE4]: [ERROR] Unreal Engine not detected
-[BuildCarlaUE4]: [Batch params]: --launch --all
--launch
--all
ECHO is off.
ERROR: The system was unable to find the specified registry key or value.

-[BuildCarlaUE4]: [ERROR] Unreal Engine not detected

Does anybody have any idea to solve problem?