carla-simulator/carla #2777: PTV-Vissim Co-simulation problem

kkyu725, 2020-04-22:
Hello
I'm trying to get running Carla in windows 10 and co-simulating with PTV-Vissim 11.00.
I have followed the instructions here: 
https://carla.readthedocs.io/en/latest/adv_ptv/

While I run the PTV -Vissim co-simulation, I got errors.
 "vissim type xxxxxx unknown. No vehicle will be spawned in carla" 

I find that vissim DS API always generate a new vehicle_id "0" when a vehicle drives across the link/connector but no vehicle spawned in vissim . That cause the set of vissim vehicle_ids be mess up. 
` 
        vehicles = {}
        for i in range(num_vehicles.value):
            vehicle_data = traffic_data[i]

            vehicles[vehicle_data.VehicleID] = VissimVehicle(
                vehicle_data.VehicleID, vehicle_data.VehicleType, vehicle_data.ModelFileName,
                vehicle_data.color,
                [vehicle_data.Position_X, vehicle_data.Position_Y, vehicle_data.Position_Z],
                [vehicle_data.Orient_Pitch, vehicle_data.Orient_Heading, 0.0], vehicle_data.Speed,
                vehicle_data.TurningIndicator)
`

jy723, 2020-04-24:
You can use vissim2020 and DrivingSimulatorProxy.dll,or change 
vissim_simulation.py class VISSIM_Veh_Data(Structure)

joel-mb, 2020-04-28:
Hi @kkyu725, did the solution proposed by @jy723 solve your problem?
Thank you!

kkyu725, 2020-04-30:
Yes, thanks so much~!

closed 2020-05-04 by joel-mb

AloneMusk, 2024-01-03:
@kkyu725 @jy723 what changes do we necessarily have to make to the VehData class in order to rectify this issue? Vissim 2020 seems to run smoothly but Vissim 2023 crashes whilst displaying the same error.

firas-elhag, 2024-05-06:
@AloneMusk were you able to solve this error?