carla-simulator/carla #2779: Build on clang greater than 8?

mrwesleycrusher, 2020-04-23:
I am trying to build Carla 0.9.8 on arch linux, and I have clang 10 installed, but Carla requires clang 8.
Is there a reason clang 8 is required? Can it not be built on newer versions of llvm?

nsubiron, 2020-04-24:
The server-side is statically linked to Unreal Engine, so it needs to compile with the same toolchain, llvm-8. The client-side can be built with any toolchain, but for consistency everything is built with clang-8.

But for instance in the [CppClient Example](https://github.com/carla-simulator/carla/tree/master/Examples/CppClient) we build the client-side C++ API with gcc.

mrwesleycrusher, 2020-04-29:
Ahhhhhh I see. I built Unreal from source though without issue. Also, I am not sure if I need to submit a seperate issue for this, but the link to the unreal build documentation points to the official Unreal Engine Wiki, which no longer exists (Epic Games' git repo README has the same issue). The instructions I followed were located at https://github.com/EpicGames/UnrealEngine/blob/release/Engine/Build/BatchFiles/Linux/README.md

nsubiron, 2020-04-30:
UE4 now downloads and uses its own copy of clang/includes/etc regardless of what you have installed. It's in "Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/...".

closed 2020-06-26 by germanros1987