carla-simulator/carla #2781: Memory leak in traffic manager

zhejun-zhang, 2020-04-24:
It seems traffic manager is not correctly cleaned up after actors are destroyed and world is reloaded. The issue is found on version 0.9.9 and 0.9.8. Version 0.9.6 which does not have traffic manager handling autopilot works fine.

To reproduce
1. run docker of (0.9.9, 0.9.8 or 0.9.6) on ubuntu 18.04. I use singularity so
 `singularity exec --nv carla_0.9.9.sif /bin/bash /home/carla/CarlaUE4.sh -benchmark -fps=10 -carla-server -opengl`
2. Activate python environment with corresponding carla egg installed.
3. Execute the minimal example below. 
4. Check memory usage with htop in the terminal. The same code runs on all three versions of carla. Memory usage is constant on 0.9.6 but increasing on 0.9.9/0.9.8

```python
import carla
import numpy as np

def spawn_vehicles(world, client, num_vehicles):
    vehicles = []
    blueprints = world.get_blueprint_library().filter("vehicle.*")
    SpawnActor = carla.command.SpawnActor
    SetAutopilot = carla.command.SetAutopilot
    FutureActor = carla.command.FutureActor

    # spawn_points
    spawn_points = world.get_map().get_spawn_points()
    np.random.shuffle(spawn_points)
    spawn_points = spawn_points[0:num_vehicles]
    # batch commands
    batch = []
    for n, transform in enumerate(spawn_points):
        blueprint = np.random.choice(blueprints)
        blueprint.set_attribute('role_name', 'autopilot')
        # batch.append(SpawnActor(blueprint, transform))
        batch.append(SpawnActor(blueprint, transform).then(SetAutopilot(FutureActor, True)))

    for response in client.apply_batch_sync(batch, do_tick=True):
        if not response.error:
            vehicles.append(response.actor_id)

    print(f'Spawned {len(vehicles)} vehicles. Should Spawn {num_vehicles}')

    return vehicles

def set_sync_mode(world, sync):
    # traffic_manager.set_global_distance_to_leading_vehicle(2.0)
    # traffic_manager = client.get_trafficmanager(8000)
    # traffic_manager.set_synchronous_mode(sync)
    settings = world.get_settings()
    settings.synchronous_mode = sync
    settings.fixed_delta_seconds = 0.1

    world.apply_settings(settings)

def run():
    for i in range(100):
        print(i)
        client = carla.Client("localhost", 2000)
        client.set_timeout(5.0)
        world = client.load_world("Town01")

        set_sync_mode(world, True)
        world.tick()

        # spawn
        vehicles = spawn_vehicles(world, client, 100)
        world.tick()

        # try the best to clean world
        client.apply_batch_sync([carla.command.SetAutopilot(x, False) for x in vehicles], True)
        client.apply_batch_sync([carla.command.DestroyActor(x) for x in vehicles], True)

        client.reload_world()
        vehicles = None
        client = None
        world = None

if __name__ == '__main__':
    run()

```
I tried to clean up as much as possible: autopilot is turned off, actors are destroyed, client is disconnected and world is reloaded. Yet memory is not freed after each iteration.
I suspect tm is the issue because if one does not touch autopilot, the memory leak will disappear. Specifically
1. While spawning, do not turn on autopilot
use 
`batch.append(SpawnActor(blueprint, transform))` 
comment out 
`batch.append(SpawnActor(blueprint, transform).then(SetAutopilot(FutureActor, True)))`
2. While cleaning, do not turn off autopilot
comment out 
`client.apply_batch_sync([carla.command.SetAutopilot(x, False) for x in vehicles], True)`

jackbart94, 2020-04-27:
Hi,

Thanks for reporting this issue! If you are worried about why TM stays alive after a map change/reload, that's an intended behavior. Since the client doesn't shut down automatically after a map change, the TM doesn't as well. Let me know if this solves your doubts!

zhejun-zhang, 2020-04-27:
Thanks for your answer! Why memory used by TM is not freed after the actor is destroyed? This prevents TM being used to benchmark autonomous agent or collect dataset where actors are respawned and map is reloaded in each episode. Or is there a way to shut down client from the pythonAPI?

jackbart94, 2020-04-27:
This is indeed an issue, we are currently changing the architecture of TM for 0.9.10 to allow for better memory cleanup, among other things. If it's a benchmark you want, maybe it's better to shut down the client anyway, although I can understand the problem with it.

zhejun-zhang, 2020-04-30:
Thanks! I’m looking forward to the new release.

closed 2020-04-30 by zhejun-zhang

davebiagioni, 2020-06-18:
First of all, thanks @jackbart94 folks for confirming this is an issue and aiming for a fix in the next version!

I know this issue is closed, but I wanted to shed a little more light on the severity of the problem, especially for users trying to run many scenarios in a persistent world / server.

I ran the script posted by @zhejun-zhang above (with 10 iterations rather than 100) and wrapped it in a call to `memory-profiler` to better quantify what's happening.  The results using 0.9.6 (top plot) versus 0.9.8 (bottom plot) are shown below.

[image]

[image]

For additional context, we're developing a multiagent RLLIB gym environment for carla 0.9.8, but have been seeing very large objects appearing in RLLIB's object store that it is unable to free.  Seems that this is a likely suspect!

jackbart94, 2020-06-18:
Hi,

Thanks for the amazing visualization! Yes this is an issue, but we've recently released T.M. 2.0 which is merged in master and included in 0.9.9.3. By any chance, would you mind checking if this problem you're having is still happening?

davebiagioni, 2020-06-19:
Yes, will try asap

davebiagioni, 2020-06-22:
This is using 0.9.9.4 ... it looks a tiny bit better in terms of the final memory footprint, but clearly the problem persists?

[image]

yasser-h-khalil, 2020-11-12:
Has this issue been solved?

totalmayhem96, 2021-01-23:
I think the GPU memory issue still persists in 0.9.10.1. Running a batch of routes and GPU usage slowly increases over time. I can get through about 20 routes before i get the "CUDA is out of memory" error... Any suggestions to minimize the impact?

memolo99, 2024-04-10:
I'm having this issue training a DQN agent on different scenarios in CARLA 0.9.12? Any suggestions?