carla-simulator/carla #2783: carla 0.8.4 driving benchmark_exemple [winerror 10045]

iron8kid, 2020-04-24:
windows 10  **  Carla 0.8.4  **python 3.5 
while trying to execute driving_benchmark_example.py 
I got this error 
**ERROR: (localhost:2000) failed to read data: timed out
ERROR: (localhost:2000) failed to read data: [WinError 10054] An existing connection was forcibly** 
ps: I tried different ports but it doesn't work
 
**PythonClient>**python driving_benchmark_example.py -v -db
INFO: listening to server localhost:2000
 WARNING: running the basic driving benchmark, to run for CoRL 2017 experiment suites, you should run python driving_benchmark_example.py --corl-2017
DEBUG: (localhost:2000) connected
DEBUG: (localhost:2001) connected
DEBUG: (localhost:2002) connected
DEBUG: STREAM b'IHDR' 16 13
DEBUG: STREAM b'bKGD' 41 6
DEBUG: b'bKGD' 41 6 (unknown)
DEBUG: STREAM b'pHYs' 59 9
DEBUG: STREAM b'tIME' 80 7
DEBUG: b'tIME' 80 7 (unknown)
DEBUG: STREAM b'IDAT' 99 8192
DEBUG: STREAM b'IHDR' 16 13
DEBUG: STREAM b'pHYs' 41 9
DEBUG: STREAM b'iCCP' 62 310
DEBUG: iCCP profile name b'Photoshop ICC profile'
DEBUG: Compression method 0
DEBUG: STREAM b'iTXt' 384 16722
DEBUG: STREAM b'cHRM' 17118 32
DEBUG: STREAM b'IDAT' 17162 530157
DEBUG: STREAM b'IHDR' 16 13
DEBUG: STREAM b'bKGD' 41 6
DEBUG: b'bKGD' 41 6 (unknown)
DEBUG: STREAM b'pHYs' 59 9
DEBUG: STREAM b'tIME' 80 7
DEBUG: b'tIME' 80 7 (unknown)
DEBUG: STREAM b'IDAT' 99 8192
INFO: START
DEBUG: (localhost:2001) disconnecting
DEBUG: (localhost:2002) disconnecting
DEBUG: (localhost:2000) disconnecting
ERROR: (localhost:2000) failed to read data: timed out
DEBUG: (localhost:2000) connected
DEBUG: (localhost:2000) disconnecting
ERROR: (localhost:2000) failed to read data: [WinError 10054] An existing connection was forcibly closed by the remote host
DEBUG: (localhost:2000) connection attempt 1: [WinError 10061] No connection could be made because the target machine actively refused it
DEBUG: (localhost:2000) connection attempt 2: [WinError 10061] No connection could be made because the target machine actively refused it
DEBUG: (localhost:2000) connection attempt 3: [WinError 10061] No connection could be made because the target machine actively refused it
DEBUG: (localhost:2000) connected
DEBUG: (localhost:2001) connected

mhmdgad97, 2020-06-07:
I have the same problem. did you figure it out?

germanros1987, 2020-06-26:
@felipecode could you help with this?

iron8kid, 2020-06-30:
> I have the same problem. did you figure it out?
> Also, I've found that when I run the benchmark script, Carla uses 100% of the hard disk bandwidth and Carla becomes "not responding" so I think this might be the problem, but I don't know why it uses this much disk bandwidth.

I solved it by changing the time_out

closed 2020-09-05 by stale[bot]

fjzs, 2023-07-18:
is there any formal solution to this issue?

AwesomePaneer, 2025-01-26:
has there been any general solution to this?