carla-simulator/carla #2785: Carla_0.9.8_Ground plane paramters

AhmedSoror, 2020-04-24:
Hi,
I would like to get the parameters (orientation) of the ground plane normal in order to find the equation of the ground plane. 

Thanks in advance

Axel1092, 2020-04-27:
Hi @AhmedSoror, do you mean the normal vector at any point of the road? You can get the `Transform` object from a `Waypoint`. And from this you can obtain the `Rotation` of the road in that waypoint. `waypoint.transform.rotation` and extract the pitch, yaw and roll parameters.
If this doesn't solve your problem let me know what you are trying to do in order to help you.

AhmedSoror, 2020-04-27:
@Axel1092 Thanks for your reply, indeed i want to get the normal vector at any point on the road within a specific region of course. I looked at the `Waypoint` you mentioned and i understood that it is a point lying on the road and points to another possible one - The vehicle could follow such positions and it will act as if autopoilot is true - thus i want to find the normal to such point.

Axel1092, 2020-04-28:
Actually the `waypoint` object gives you the position and orientation of the road at a given point. Therefore, from the `yaw`, `pitch` and `roll` in `waypoint.transform.rotation` you can obtain the normal vector of the road. Here you have a little pseudo-code example of how to do this:
```
cy = cos(yaw)
sy = sin(yaw)
cr = cos(roll)
sr = sin(roll)
cp = cos(pitch)
sp = sin(pitch)
normal_vector = carla.Vector(cy*sp*sr-sy*cr, sy*sp*sr+cy*cr, -cp*sr)
```
Don't forget to convert the angles to the right units (radians or degrees).

AhmedSoror, 2020-04-29:
@Axel1092  It worked. I used the waypoint vector as you mentioned and rotated it with 90 about the vehicle axis using its rotation and it works. Thanks for your support

closed 2020-06-26 by germanros1987