carla-simulator/carla #2789: Traffic Manager hangs after reload

dianchen96, 2020-04-27:
Hello,

In the lasted version (0.9.9), `client.get_trafficmanager()` will hang if the world is reloaded and a traffic manager has been created with sync mode before. A minimal example to reproduce this is 

```
import carla
client = carla.Client('127.0.0.1', 2000)
world = client.load_world('Town01')

print ("Loaded world")
tm = client.get_trafficmanager()
tm.set_synchronous_mode(True)
print ("TM set to sync")
world = client.load_world('Town01')
print ("Loaded world")
tm = client.get_trafficmanager() # This will hang
print ("Got TM")
```

I am using Python 3.7 and Ubuntu 18.04.

jackbart94, 2020-04-27:
Hi,

Thanks for reporting the issue! This is indeed happening (even though you are slightly misusing the commands) because the map change/reload happens with synchronous mode on. You should briefly deactivate it before the map change/reload, then reactivate it. Let me know if this solves your issue!

dianchen96, 2020-04-28:
Thanks for your quick response! Setting sync mode to False indeed solves my problem. Also it seems like the CARLA server should be launched without the `-benchmark` flag for it to work.

filangel, 2020-05-09:
@jackbart94 I have reproduced the issue, however, by setting the traffic manager to async mode (i.e., `tm.set_synchronous_mode(False)`) before reloading the world does *not* fix the issue for me!

TimLued, 2020-07-23:
Hello,

unfortunately Carla still crashes after a few successful loops (in my case 4) with the following error:
RuntimeError: failed to connect to newly created map

```
while 1:
	tm = client.get_trafficmanager()
	tm.set_synchronous_mode(True)
	print ("TM set to sync")

	settings = world.get_settings()
	settings.fixed_delta_seconds = 0.1
	settings.synchronous_mode = True
	world.apply_settings(settings)
	print("Carla sync mode ON")

	# do stuff ...

	tm.set_synchronous_mode(False)
	world.apply_settings(carla.WorldSettings(synchronous_mode=False))
	world = client.load_world('Town01')
	# after some successful loops ... RuntimeError: failed to connect to newly created map
	print ("Loaded world")
```

adscib, 2020-08-14:
I have the same problem. Tried all combinations of setting the world and/or traffic manager to sync/async mode before loading the new world, but the traffic manager always hangs after.
@jackbart94 can we reopen this?

doterop, 2020-08-14:
Hi @TimLued @adscib ,

Can you confirm the exact version of Carla that are you using?

filangel, 2020-08-14:
I can reproduce @adscib's issue in **CARLA 0.9.9.4**!

TimLued, 2020-08-14:
The error occurs for me in Carla 0.9.9.4

adscib, 2020-08-15:
I was also running CARLA 0.9.9.4 from the official pre-compiled release running on Ubuntu 20.04 with Python 3.7.3. I had a different issue with the traffic manager in CARLA 0.9.9 so didn't get that far, but I can check if the same issue occurs there if that would be helpful.

dianchen96, 2020-08-28:
I am reopening this because I find the solution **does not work** if there are actors spawned before closing the reloading the world. This happens to me at 0.9.9.4

Beskhue, 2020-09-01:
I'm encountering the same issue on Carla 0.9.9.4.

Interestingly, if I restart the Python script, it reconnects correctly. However, dropping the `client` handle and creating a new one between connection does not work.

adscib, 2020-09-05:
I can also confirm this issue is still present in the 0.9.10 pre-release and in the nightly build I downloaded yesterday.

yjiang05, 2020-09-08:
I also encounter the same issue in CARLA 0.9.9.4 and I use Ubuntu 20.04 and Python 3.7. 

What I've tried and found:
1. Setting TM's sync mode to `False` either before or after reloading the world will cause the TM to hang.
2. Leaving the server running and restarting the script, it works well (similar to @Beskhue's description).
3. Starting two servers with different ports, using the same code to create two clients, the first works but the second client's TM will hang (similar to https://github.com/carla-simulator/carla/issues/3220#issuecomment-685976275).
4. Without TM, everything works fine.

Beskhue, 2020-09-09:
When dropping the client, some network connections are kept open. This is noted in #3212. When these connection are completely dropped the script can successfully connect again, even when using traffic manager and spawning agents.

A workaround is to create the Carla client inside a new process (from multiprocessing). When this process completes, the client and all connections are dropped. Now, launching a new process and setting up Carla again will succeed.

For example:

```python
from multiprocessing import Process
import carla

def run():
    client = carla.Client("localhost", 2000)
    world = client.load_world("Town01")

    traffic_manager = client.get_trafficmanager()
    traffic_manager.set_synchronous_mode(True)

    settings = world.get_settings()
    settings.fixed_delta_seconds = 0.1
    settings.synchronous_mode = True
    world.apply_settings(settings)

    # ... Do things ...

    traffic_manager.set_synchronous_mode(False)

    settings = world.get_settings()
    settings.fixed_delta_seconds = 0.0
    settings.synchronous_mode = False
    world.apply_settings(settings)
    
while True:
    p = Process(target=run, args=())
    p.start()
    p.join()
```

See also: https://github.com/carla-simulator/carla/issues/3212#issuecomment-686979191

yjiang05, 2020-09-10:
Hi @Beskhue,

Thanks for your comment! It works when `multiprocessing` is used. However, in my case, after `close()`, which closes a client, in any process being called, if I hope to re-instantiate (re-create) a client in the same process, the connection will fail.

So, does it imply that (1) I should create "sub-process" within each process (though I don't know if it is possible) to re-instantiate a client after closing a previous one OR (2) I should never call `close()` to close a client within a process?

And also here is a code snippet I used to close a client:

```Python
def close():
    # clean up actors, sensors, etc

    traffic_manager.set_synchronous_mode(False)

    settings = world.get_settings()
    settings.fixed_delta_seconds = 0.0
    settings.synchronous_mode = False
    world.apply_settings(settings)
```

Beskhue, 2020-09-11:
@yjiang05 Neither (1) or (2). Using this workaround, you do not re-use the processes. You should create a process per client setup and teardown. So, you create a process, and in it you setup the client, run the things you want to run, and close the client. To run another thing, you repeat the same steps.

jackbart94, 2020-09-15:
Sorry if it took so much to open a PR for this issue, it appears we have found a way to fix it!

It will soon be merged in `dev` and then to `master` when 0.9.10 comes out, but if you want to test it (and give us feedback!) the branch is `traffic_manager/map_change_fix`

Thanks for your patience!

closed 2020-09-16 by bernatx

belorenz, 2021-01-15:
This issue still exist for me. For everybody who is looking for a workaround, you can simply create another tm with a different port.
For example if you created the TM in the first map with port 8000, just create another one with port 8001. 
Not ideal, but worked for me.

PR #3288 (2020-09-16): Fixes a map change error when TM is in synchronous mode
#### Description

An erroneous update of the TM while in synchronous mode was causing the client to hang on map changes.

Fixes #3220
Fixes #2789

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04, Windows
  * **Python version(s):** Python 3.6
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks

Seems to not be working for Windows, further testing is required.

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3288)