carla-simulator/carla #2790: IMU measurements do not reflect the rotation of the IMU relative to the vehicle

ItsTimmy, 2020-04-27:
**Issue**: IMU measurements are always reported with the axes aligned with the vehicle, even if the IMU is attached with some rotation.

**To reproduce**:

  1. Start the Carla server
  2. Spawn an ego vehicle using the `examples/manual_control.py` script
  3. Run the following code in a python 3 console:

```
import carla

rolename = "hero"

client = carla.Client("localhost", 2000)
world = client.get_world()
hero = next(filter(lambda a: a.attributes.get('role_name') == rolename, world.get_actors()).__iter__())

def imu_callback(name, data: carla.IMUMeasurement):
    print("{}, frame {}, accel: ({}, {}, {})".format(name, data.frame_number, data.accelerometer.x, data.accelerometer.y, data.accelerometer.z))

sensors = []

for i in [0, 1]:
    bp = blueprint = world.get_blueprint_library().find('sensor.other.imu')
    imu_rolename = "imu{}".format(i)
    bp.set_attribute('role_name', imu_rolename)
    bp.set_attribute('sensor_tick', '0.01')
    transform = carla.Transform(carla.Location(0, 0, 0), carla.Rotation(i * 10, i * 20, i * 30))
    sensor = world.spawn_actor(bp, transform, attach_to=hero)
    sensors.append(sensor)

sensors[0].listen(lambda data: imu_callback("imu0", data))
sensors[1].listen(lambda data: imu_callback("imu1", data))
```

This will attach 2 IMUs at different orientations to the `here` vehicle, and continuously print out the measurements from each IMU. Observe the Z values: Both will be +1G, despite the fact that `imu1` is attached to the `hero` vehicle at a different angle from `imu0`.

ItsTimmy, 2020-04-27:
After doing some experiments, and driving around and watching the values, I am now more confused. It seems as though the acceleration of the vehicle is correct with respect to the orientation of the IMU, but acceleration due to gravity is not. It seems like there is always a 1G acceleration added to the Z axis, but the actual acceleration of the car may be correct otherwise.

I would need to understand more about the inner workings of how the IMU measurements are taken in order to proceed. I welcome the input of anyone who has experience with this.

ItsTimmy, 2020-04-27:
I have found the issue [here](https://github.com/carla-simulator/carla/blob/1a2c2ea76aff4b284dd69fbcbe26b95846d6c93f/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/InertialMeasurementUnit.cpp#L125-L130). `GRAVITY` should be added to the Z component before rotating the vector.

I am currently having issues building Carla from source on my machine, so I cannot yet implement and test this fix.

closed 2020-05-06 by marcgpuig

a4zhangfei, 2020-05-20:
hi,blueprint ' sensor.other.imu 'not found, why don't I have IMU?
There is no sensor in my blueprint other.imu and sensor.other.radar

PR #2795 (2020-05-06): Fixed IMU gravity
#### Description

On the IMU sensor, there was an issue where acceleration due to gravity was added after rotating the actor acceleration vector to the orientation of the IMU. This meant that there was a constant 9.8 m/s^2 added to the Z axis of the accelerometer readings regardless of the orientation of the accelerometer. This PR changes this so that gravity is added to the Z axis before rotation.

Fixes #2790 

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 2.7, 3.6
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbac [... cut]