carla-simulator/carla #2792: The C++ compiler "/usr/bin/clang++-8" is not able to compile a simple test program

revikiran1991, 2020-04-27:
I am trying to compile Carla 0.9.9 on ubuntu18.04:
I get the following error:
```Setup.sh: Compiling libc++.
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- The ASM compiler identification is Clang
-- Found assembler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- broken
CMake Error at /usr/share/cmake-3.10/Modules/CMakeTestCXXCompiler.cmake:45 (message):
  The C++ compiler

    "/usr/bin/clang++-8"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/user_name/carla/Build/llvm-8.0-build/CMakeFiles/CMakeTmp
    
    Run Build Command:"/usr/bin/ninja" "cmTC_7869f"
    [1/2] Building CXX object CMakeFiles/cmTC_7869f.dir/testCXXCompiler.cxx.o
    [2/2] Linking CXX executable cmTC_7869f
    FAILED: cmTC_7869f 
    : && /usr/bin/clang++-8     CMakeFiles/cmTC_7869f.dir/testCXXCompiler.cxx.o  -o cmTC_7869f   && :
    /usr/bin/ld: cannot find -lstdc++
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:39 (project)

-- Configuring incomplete, errors occurred!
See also "/home/usr_name/carla/Build/llvm-8.0-build/CMakeFiles/CMakeOutput.log".
See also "/home/usr_name/carla/Build/llvm-8.0-build/CMakeFiles/CMakeError.log".
Util/BuildTools/Linux.mk:135: recipe for target 'setup' failed
make: *** [setup] Error 1
```
And the `CMakeError.log`:
```
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /usr/bin/clang++-8 
Build flags: 
Id flags:  

The output was:
1
/usr/bin/ld: cannot find -lstdc++
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Determining if the CXX compiler works failed with the following output:
Change Dir: /home/usr_name/carla/Build/llvm-8.0-build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/ninja" "cmTC_90d92"
[1/2] Building CXX object CMakeFiles/cmTC_90d92.dir/testCXXCompiler.cxx.o
[2/2] Linking CXX executable cmTC_90d92
FAILED: cmTC_90d92 
: && /usr/bin/clang++-8     CMakeFiles/cmTC_90d92.dir/testCXXCompiler.cxx.o  -o cmTC_90d92   && :
/usr/bin/ld: cannot find -lstdc++
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

germanros1987, 2020-06-26:
@marcgpuig could you look into this?

marcgpuig, 2020-07-03:
Hi @revikiran1991, first of all, this has nothing to do with CARLA and can be happening because of different causes.
To me, the main clue here is here:
```
/usr/bin/ld: cannot find -lstdc++
``` 
And looks like in your case the solution could be in [this post](https://stackoverflow.com/questions/2086072/linking-using-g-fails-searching-for-lstdc).
Hope this helps.

tejeshanil, 2020-08-17:
Hi @revikiran1991 
Check after installing libstdc++<ver>-dev  package.
This solved issue for me.

closed 2020-11-08 by stale[bot]

feyza-droid, 2021-08-13:
**I solved this issue by removing all gcc versions I had and installing gcc again.**

Check how many versions of gcc you have `dpkg -l | grep gcc`. Also you can look at the directory `/usr/lib/gcc/x86_64-linux-gnu`.
In my case there are 7, 7.5, 8, 9, 10, 11.
So I removed each of them by typing `sudo apt-get purge gcc-{version}`

Ex: `sudo apt-get purge gcc-7`
When you remove all of them `gcc --version` gives nothing which is fine.

Then install gcc with `sudo apt-get install build-essential`.
Check whether gcc is installed correctly by typing `gcc --version`.

Now `make PythonAPI` works fine.
Hope it helps :)

anthonymorast, 2024-02-03:
Late to the party, but [this StackOverflow answer](https://stackoverflow.com/questions/74543715/usr-bin-ld-cannot-find-lstdc-no-such-file-or-directory-on-running-flutte) saved my life in 2024

There was an issue with the latest Linux mint and the version of clang/gcc. The simplest solution was to install libstdc++12-dev. None of the other suggestions worked.

```bash 
sudo apt install libstdc++-12-dev
```