carla-simulator/carla #2793: How to set the status of a specific traffic light in the scene?

hepeng187, 2020-04-28:
I can set all the traffic lights in the scene at the same time, but I do n’t know how to set only one of the traffic lights, and how to know the position of this traffic light in the scene.

germanros1987, 2020-06-26:
@Axel1092 could you comment on this?

germanros1987, 2020-06-26:
@sergi-e could you document the outcome?

Axel1092, 2020-06-26:
Hi @hepeng187,
You can identify lights by their position which you can query in the follwoing way:
```
    client = carla.Client()
    client.set_timeout(10.0)
    world = client.get_world()
    light_manager = world.get_lightmanager()
    lights = light_manager.get_all_lights()
    for light in lights:
        position = light.location
        # do something with position and light
```
Then you can use the functions we provide to affect each light (see carla.Light in the PythonAPI: https://carla.readthedocs.io/en/latest/python_api/#carla.Light)

Axel1092, 2020-06-29:
Hi @hepeng187,
Sorry, in my previous message I was confused regarding traffic lights and street lights. About traffic light position, the traffic light is a kind of `carla.Actor` therefore you can query its position as with any other actor in the scene with `actor.get_location()`. You can get the traffic light object from a vehicle.
You can also get a traffic light object through the `Landmark` object, which encodes traffic lights and signals encoded in OpenDRIVE files however this is yet not available in the current maps we provide but it is in any custom OpenDRIVE map with traffic lights loaded into Carla with the standalone mode (https://carla.readthedocs.io/en/latest/adv_opendrive/).

hepeng187, 2020-06-30:
Hi @Axel1092 ,
I want to get a traffic light object through the Landmark object, and I have a  custom OpenDRIVE map that encodes traffic lights and signals. Meanwhile I run Carla in OpenDRIVE standalone mode.  But the function get_traffic_light always return None.  as follows：
```
        tmp_map = world.get_map()
        for landmark in tmp_map.get_all_landmarks_of_type('1000001'):
            traffic_light = world.get_traffic_light(landmark)
            print(traffic_light)
```

Axel1092, 2020-07-01:
Hi @hepeng187,
Maybe the traffic lights are not being ingested by the simulation (they should be visible in the unreal camera if they are). This may be due to the fact that the traffic light is defined within a junction but without a controller in OpenDRIVE. Currently, Carla only enables a traffic light in the simulation if the traffic light is define outside a junction or if there is a controller associated to this traffic light in order to regulate the junction.

hepeng187, 2020-07-03:
Hi@Axel1092  ,
Sorry ,I don't quite understand what you mean.  
When I play Carla ,the traffic lights are visible in the unreal camera. But when I stop traffic lights disappeared. Just like TownBig, Can the function (get_traffic_light(landmark))  be used in the TownBig virtual scene?

 The labels of my opendrive file are as follows:  Is this right？
```
     .......
    <road name="" length="38.38129668" id="42" junction="-1">
         ......
        <signals>
            <signal s="2" t="5" id="1" name="11" dynamic="yes" orientation="-" zOffset="0" type="1000001" 
                        country="OpenDRIVE" subtype="-1" value="-1"/>
        </signals>
    </road>

           .....

    <controller name="ctrl001" id="1">
        <control signalId="11" type="0" />
    </controller>

      ......
    <junction name="gneJ1" id="1">
         ......
        <controller id="1" type="0"/>
    </junction>

```

Axel1092, 2020-07-03:
Maybe this is just an example but the signal id is 1 and the controller points towards signal id 11.
The function `world.get_traffic_light(landmark)` should return the corresponding `carla.TrafficLight` object that allows you to control it. Check, that the `landmark.id` matches the id of the traffic light you need and not other signals.

hepeng187, 2020-07-07:
@Axel1092  The controller signal id has now been corrected ( the controller points towards signal id 1) , now the landmark_id has matched the id of signal, but it still return none. I'm about to collapse...

Axel1092, 2020-07-09:
Hi @hepeng187, we found that this problem was a bug with traffic lights that caused the actors to be missing on the client side. We are fixing this in the new PR currently under review. Thank you for pointing these problems, you may try to move to the branch `axel1092/traffic-light-cleanup` if you are working with master and see if the problems are solved.

closed 2020-07-10 by Axel1092

LencyGE, 2022-04-01:
Well, actually, I want to know how to set all traffic lights in a map to the same state? @[hepeng187](https://github.com/hepeng187)

PR #3041 (2020-07-10): Traffic light fixes
#### Description

Reworked traffic light system, now all traffic lights use `UTrafficLightComponent` to control their behavior (solves many determinism issues).
The recorder now only "ticks" on the posttick event in the `FCarlaEngine`.
Reworked `ATrafficLightManager` to register traffic lights from OpenDRIVE and existing lights in the editor.
Remade frozen behavior, now it affects all traffic lights in the map and it works using the `ATrafficLightManager`. A new function in carla.World has been added to implement this functionality and the `traffic_light.freeze(True)` funct [... cut]