carla-simulator/carla #2797: make launch fails due to missing members in global namespaces

MichaelHannalla, 2020-04-28:
Unreal Engine Version: 4.24 and built sucessfully
Carla Version: 0.9.X cloned on 28 - April - 2020 from master branch
I have building errors due to missing members in global namespaces, I am using clang-8 and llvm-8 to build carla. 

Please check my terminal output
[michael_carlabuilderror.txt](https://github.com/carla-simulator/carla/files/4546837/michael_carlabuilderror.txt)

Thanks in advance

kminoda, 2020-05-03:
I also got the same issue. Does anyone have any idea of how to solve this problem?

DSantosO, 2020-06-17:
I had also the same issue on Ubuntu 18.04 and for me it seemed to be a problem with the version of clang. After removing all the installed versions of clang and llvm, I reinstall them following carefully the instructions on the carla documentation: https://carla.readthedocs.io/en/latest/build_linux/ at section 'Dependencies' and then it was working. I hope this is useful.

germanros1987, 2020-06-26:
@MichaelHannalla and @kminoda, did @DSantosO feedback solve your issue?

germanros1987, 2020-07-30:
Issue closed due to lack of activity. Feel free to open a new issue if you still consider this is a critical matter.

closed 2020-07-30 by germanros1987

qhaas, 2020-09-04:
> I had also the same issue on Ubuntu 18.04 and for me it seemed to be a problem with the version of clang. After removing all the installed versions of clang and llvm, I reinstall them following [...]

Encountered the same error with CARLA 0.9.7, clang 8, and Ubuntu 18.04.  The problematic packages were `libc++-${CLANG_VER}-dev libc++abi-${CLANG_VER}-dev libc++1-${CLANG_VER} libc++abi1-${CLANG_VER}`, where CLANG_VER is the version of clang you are using (with 0.9.7, that *should* be clang-7, but I am dealing with a special use case) once removed, it clears the error

NishanthARao, 2021-05-31:
@qhaas Thanks a ton! In my case, I was using clang-8, CARLA 0.9.10 and UnrealEngine 4.24. I did ```sudo apt remove libc++-8-dev libc++abi-8-dev libc++1-8 libc++api1-8```. After that I removed the build folder in CARLA directory and ran the make commands. It worked perfectly.

StephanieMussi, 2021-09-03:
> @qhaas Thanks a ton! In my case, I was using clang-8, CARLA 0.9.10 and UnrealEngine 4.24. I did `sudo apt remove libc++-8-dev libc++abi-8-dev libc++1-8 libc++api1-8`. After that I removed the build folder in CARLA directory and ran the make commands. It worked perfectly.

I can't thank you enough for this! It solved my problem after I have been struggling for weeks! Thank you!

erosinnocenti, 2021-12-07:
> 

Same here! Thanks for the tip, worked perfectly.

Jonghwanshin, 2022-01-18:
I would like to share my experience for someone who experience this issue.

I was stucked in this issue for several days. 
Then I saw this page then I checked if there is any libc++ installation with `dpkg -l | grep libc++`.
I could see there are some libc++ packages in my computer like this.

```
$ dpkg -l | grep libc++

ii  libc++-dev:amd64                                6.0-2                                           amd64        LLVM C++ Standard library (development files)
ii  libc++-helpers                                  6.0-2                                           all          LLVM C++ Standard library - build helpers
ii  libc++1:amd64                                   6.0-2                                           amd64        LLVM C++ Standard library
ii  libc++abi1:amd64                                6.0-2                                           amd64        LLVM low level support for a standard C++ library
```

I could figured out this library was linked in the building process instead of libc++ in CARLA build folder, 
so I removed by `sudo apt purge libc++*` and now it works.