carla-simulator/carla #2799: get acess to actorblueprint which spawned by hand in UE4 editor

hubing1994, 2020-04-29:
Hellow everyone:
      I want to spawn some vehicles by hand in UE4 Editor，I notice that there are some blueprint in /Game/carla/blueprint/vehicles。I can spawn a AudiA2 in UE4 editor。however，in the simulation，I can not get acess to the "AudiA2" ，What can I do ？

mrwesleycrusher, 2020-04-29:
I would recommend adding your blueprint into CARLA by following the instructions here: https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/

hubing1994, 2020-04-30:
I don't want to creat a new vehicle blueprint. I just want to use the exised vehicle blueprints such as "AudiA2". I try to drag the "AudiA2" blueprint into the UE4 editor and run the simulation. there is a AudiA2 in the game window, However, I cannot get the "AudiA2" by world.get_actors()

mrwesleycrusher, 2020-04-30:
So by not spawning it in CARLA, the world doesn't know about the actor, because it isn't a CARLA actor

mrwesleycrusher, 2020-04-30:
You should just spawn it via spawn_actor

closed 2020-06-26 by germanros1987