carla-simulator/carla #2800: More ergonomic drive cycle simulation generation in CARLA

mrwesleycrusher, 2020-04-29:
Hey maintainers!
I hope that this issue finds you well, I want you to know I am only complaining and begging for features because I really want the simulator to continue to be the most ergonomic of the Unreal-based autonomous vehicle simulators.

So I went through the backlog and found all these issues that I think are related:
#2542 
#2563 
#2763 
#2746 

The way that interaction is done with OpenDRIVE in the CARLA examples gives you an idea of how to start to interact with OpenDRIVE, but it is difficult to create paths for drive cycles that vehicles can follow. In my experience it requires a lot of hard-coding in road ids and other OpenDRIVE metadata to be able to create these paths that non-ego vehicles can follow in the simulator. It would be really great if there was a way to even just drive a vehicle using manual_control, and have the recorder record all the OpenDRIVE waypoints the vehicle touches during the drive cycle, so that on these beautiful maps there is a way to create a specified, repeatable path quickly.

Another feature that I would like to request is being able to run CARLA in a sort of "annotation mode" where all OpenDRIVE metadata for the map is displayed so that even if the hard-coding is still necessary, we aren't fighting the lane explorer's random walks trying to get information for the roads we want.

Thanks!

EliteTUM, 2020-05-02:
@mrwesleycrusher : You could try to record the waypoints closest to the current location of the vehicle and record them.

germanros1987, 2020-06-26:
@mrwesleycrusher I think your initial hypothesis is wrong. You don't need to provide OpenDRIVE coordinates/info to create trajectories. You can record your trajectory using a modification of the manual_control (for instance), by simply recording the Waypoints.

You can also use the BasicAgent to send your vehicle to a given point, or even the functionality provided by Scenario_Runner. You definitely don't need to record OpenDRIVE data.

Regarding the "annotation mode"... what kind of metadata do you want? This sounds like the non_rendering_mode script (in the examples folder). Have you tried that?

closed 2020-07-30 by germanros1987