carla-simulator/carla #2801: LowLevelFatalError about Vulkan

lemonriver, 2020-04-30:
**Error details:**

4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.

LowLevelFatalError [File:Unknown] [Line: 772]

VulkanRHI::vkQueueSubmit(Queue, 1, &SubmitInfo, Fence->GetHandle()) failed, VkResult=-4
 at /home/xxxx/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanQueue.cpp:71
 with error VK_ERROR_DEVICE_LOST

Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=124960 LargeMemoryPoolOffset=256096
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

./CarlaUE4.sh: line 5: 98932 Segmentation fault      "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 $@

eloy96vg, 2020-04-30:
I solved vulkan related problems running "sudo vulkaninfo" on console before initializing the program

closed 2020-06-26 by germanros1987

tinmodeHuang, 2020-08-20:
> I solved vulkan related problems running "sudo vulkaninfo" on console before initializing the program

hey @eloy96vg 
I also tried to install vulkan for my device but some error occurred, which you can refer to #3196 , I'll appreciate it if you will give me some tips or your used way of installling vulkan

Kin-Zhang, 2022-05-17:
for computer which have `sudo: vulkaninfo: command not found`
following:
```bash
sudo apt-get install vulkan-utils
```