carla-simulator/carla #2803: How debug in LibCarla Server with IDE (e.g. VS Code)?

EliteTUM, 2020-04-30:
Hello,

I found [this issue](https://github.com/carla-simulator/carla/issues/2747) on how to debug the client, but I do not manage to debug into the LibCarla server code.

I work with Ubuntu 18.04, Carla 0.9.9 and VS Code.

I tried changing the Makefile for Linux such that with `make launch` and `make package` the debug build types are built. But neither starting with `./CarlaUE4.sh` and attaching nor directly starting from VSCode worked for me. Strangely, the debug symbols for UnrealEngine are recognized (e.g. I can step through `PrimitiveSceneInfo.cpp`), but not for CARLA.

Can someone provide some guidance, on how to proceed here?

Thank you!

EliteTUM, 2020-05-05:
Ok, solved it. Simply added -g to the server CXX compile flags. Now I can both launch and attach without any problems to my local build.

closed 2020-05-05 by EliteTUM

gengtsh, 2022-04-06:
hi, can you please let me know how you config vs and how to debug libcarla? thanks

eyabesbes, 2024-04-29:
> Hello

Hi,

I'm also working on Ubuntu 18.04 and Carla (0;9.12) and I want to run the file codes with the PythonAPI folder on VSCode.
So far, I have  installed carla and Anaconda. Then, I have created a virtual environment (carla-sim)  using anaconda in which I installed all the required libraries.
Now, when I was trying to debug when of the code files provided within PythonAPI folder using the anaconda environment that  I have created, I get the following error:
[image]
do you have any idea on how can I resolve it ?

donghaiwang, 2024-07-18:
> Ok, solved it. Simply added -g to the server CXX compile flags. Now I can both launch and attach without any problems to my local build.

Can you describe your debugging configuration steps in detail? Thank you