carla-simulator/carla #2805: Create a new map or Map customization

EugenM121, 2020-05-01:
Hallo Carla,
I see a lol of such questions, but can not find answer. My carla vesion is 0.9.8, My OS- Win10. Now I want to create a new map. But i can not get RoadRunner(wait about two weeks).

Is it possible to create a new map(Advanced highway Ai and Road System) in CARLA
using an Unreal Engine without using Roadrunner??? If it is possible, are instrucktions the same??

I want to customize existing maps. I need to remove every thing from the map except roads and grounds fro spawning vehicles. I went through https://carla.readthedocs.io/en/0.8.4/map_customization/ this but couldn't get this. Can you maybe write a small example?

Thank you)

germanros1987, 2020-06-26:
@EugenM121 as long as you are able to provide an OpenDRIVE file along with your FBXs the CARLA simulator will be able to ingest it. Do you have a way to create an OpenDRIVE file that represents your map?

I guess the answer is "no". 

We should probably provide a tool in UE4 to produce/edit basic OpenDRIVEs... @XGodina @marcgpuig ?

zhoushiyang12, 2020-06-29:
Hallo Carla,
I fllowed the Tutorials  https://carla.readthedocs.io/en/latest/tuto_A_add_map/
but when 

./build.sh map_file # map_file has no extension

I encounter "Limited Authority"  problem. So i  execute " source ./build.sh map_file",  but stil  encounter a problem like this :

XODR file doesn't exist, ignoring crosswalks from openDRIVE
Loading...
zsh: segmentation fault (core dumped)  ./RecastBuilder "$1.obj"

what's wrong? could you help me.

germanros1987, 2020-06-29:
@bernatx could you please follow up on this?

MichaelKlin, 2020-06-30:
Hi @germanros1987 i use unreal engine 4 to change your map und create new map. I have a small question. Can i add my Map in Release on Carla??

closed 2020-09-05 by stale[bot]