carla-simulator/carla #2806: [Windows] project stoppts to load or fails to launch after build

MichaelKlin, 2020-05-02:
I followed all the instructions at "how_to_build_on_windows/ " very very closely.
I am trying to build Carla 0.9.9 (latest) with "make launch" using unreal 4.22.2 and Visual Studios 17.
I have all prerequisites installed (make, git, python 3.6 (x64)).
I am using the x64 Native Tool Command Prompt for Visual Studios 2017
The build itself is completed successfully, but when opening the project with the unreal editor I receive the message

[image]
[image]

If Editor opens, i can not see maps and other. What should i do??
Thanks you

MichaelKlin, 2020-05-03:
Resolved

closed 2020-05-03 by MichaelKlin

VincentDupontAB, 2020-05-05:
Could you describe what was your process to resolve this issue?

MichaelKlin, 2020-05-07:
@VincentDupontAB there were so many mistakes. I forget, what I have done. But i am sure, that in my projeckt was not such files like  ../static/road/.. . And i downloaded new project-zip file. And founded that files. and added to my projeckt