carla-simulator/carla #2807: Missing Semantic Segmentation labels in OpenDrive generated maps

FrancescoPinto, 2020-05-03:
Hi,
I'm writing to point out that the semantic segmentation is not working properly with maps generated from .xodr files (Carla 0.9.9 in Ubuntu 18.04). Indeed, the road/sidewalks labels are not assigned to the road/sidewalks mesh (while they are to other objects). I guess a potential workaround could be to iterate, in the Tagger.cpp TagActor method, over the actors with naming (Actor_{i} where i represents and index i=0,1,...) and to assign them the appropriate labels. Though, it's not the cleanest solution, and I'm not sure it can make a distinction between sidewalks and roads.

germanros1987, 2020-06-26:
@marcgpuig @Axel1092 this issue has your name on it.

marcgpuig, 2020-07-01:
Hi @FrancescoPinto, this is in our roadmap and it will be addressed soon! Hopefully for the next release `0.9.10` or `0.9.11`. We also need to investigate how to properly do it, so if you make some progress make sure to report it here, it will be helpful, thanks!

FrancescoPinto, 2020-07-04:
Hi @marcgpuig , I implemented a temporary fix in this [branch](https://github.com/FrancescoPinto/carla.git)

marcgpuig, 2020-07-06:
That's great! We will take a look at your fork when we push for it. Thanks!

closed 2020-09-12 by stale[bot]