carla-simulator/carla #2808: COOKED content error in Carla 0.9.9 on windows

MichaelKlin, 2020-05-03:
I have installed UnrealEngine_4.24 and Carla 0.9.9 on my Windows 10 system. During make I did not got any error. When I run "\carla\Unreal\CarlaUE4\Binaries\Win64\CarlaUE.exe" I got below error:-
The global shader cache file 'Program Files/Epic Games/UnrealEngine_4.24/Engine/GlobalShaderCache-PCD3D_SM5.bin' is missing.
Your application is built to load COOKED content. No COOKED content was found; This usually means you did not cook content for this build.
It also may indicate missing cooked data for a shared plateform(e.g., OpenGL under Windows): Make sure your platform's packaging settings include this Targeted RHI.

Alternatively build and run the UNCOOKED version instead.

Please let me know what needs to be done in this. @sergi-e
Thanks for helping!

superying, 2020-05-26:
I have the same issue. Any solution?

convenite, 2020-06-12:
It seems a common issue with Unreal Engine. Now I'm reading 
"https://forums.unrealengine.com/development-discussion/c-gameplay-programming/2105-globalshadercache-pc3d_sm5-bin-is-missing"

germanros1987, 2020-06-26:
@sergi-e could you provide guidance on this issue and probably update the documentation?

sergi-e, 2020-06-29:
Hi, @MichaelKlin! 

It would be great to know a bit more about how did you made the build. 
Did you downloaded the assets that appear in: `\Util\ContentVersions.txt`? If not, do so and extract them in `Unreal\CarlaUE4\Content\Carla`.

EsquilliDev, 2020-09-04:
I'm having the same issues and yes, I have downloaded and extracted the assets to that folder.

closed 2020-11-14 by stale[bot]

delenius, 2020-12-18:
I am also seeing this issue when running with the `-opengl3` flag, in a Linux VM/Mac OS host (I cannot run OpenGL 4 in this configuration), using the tar.gz bundle, release 0.9.9. This appears to be the same problem as #1253.

zjzjg1280, 2021-11-27:
Has anyone solved this problem?