carla-simulator/carla #2812: Synchronous mode: (non-)deterministic results

Vaan5, 2020-05-04:
## Intro
I am running a simple script in synchronous mode which does the following:
* spawn an ego vehicle
* in a loop do:
  * set velocity
  * initiate tick
  * wait to receive data 
  * store it.

At the end of the script I store the received results and compare them with reference data.

Here is the code (i've simplified the synchronous mode example):
```python
import carla
import queue
import numpy as np

snapshots = queue.Queue()

def push_snapshot(snapshot):
    snapshots.put(snapshot)

def write_res(results):
    with open("results.txt", "w") as f:
        for t, res in results:
            f.write("{:20}\t{:20}\t{:20}\t{:20}\n".format(t, res[0], res[1], res[2]))

def compare_res():
    with open("results.txt", "r") as f:
        with open("reference.txt", "r") as r:
            print("Results {}".format(f.read() == r.read()))

def get_position(frame_id, vehicle_id):
    snapshot = snapshots.get(timeout=2)
    assert(frame_id == snapshot.frame)
    transform = snapshot.find(vehicle_id).get_transform()
    return transform.location.x, transform.location.y, transform.location.z

def main():
    sim_time = 0.0
    sim_time_end = 20.0
    delta_time = 0.05
    results = []
    index = 0

    velocity = [0.0] * 80 + [2.0] * 160 + list(np.linspace(2.0, 0.0, 120)) + [0.0] * 40

    client = carla.Client("127.0.0.1", 2000)
    world = client.get_world()

    # set mode to synchronous
    frame_id = world.apply_settings(carla.WorldSettings(True, False, delta_time))
    print("Synchronous mode will be applied in frame: {}".format(frame_id))

    # spawn car
    spectator = world.get_spectator()
    blueprint_library = world.get_blueprint_library()
    map_ = world.get_map()
    vehicles = blueprint_library.filter("vehicle")
    bp = vehicles[0]
    vehicle = world.spawn_actor(bp, map_.get_spawn_points()[0])
    world.tick()    # just so that I see it in the simulator

    world.on_tick(push_snapshot)

    while sim_time <= sim_time_end:
        vehicle.set_velocity(carla.Vector3D(y=velocity[index]))
        # client.apply_batch_sync([carla.command.ApplyVelocity(vehicle.id, carla.Vector3D(y=velocity[index]))])
        print("before tick")
        frame_id = world.tick()
        print("after tick {}".format(frame_id))
        sim_time += 0.05
        results.append((sim_time, get_position(frame_id, vehicle.id)))
        index += 1

    world.apply_settings(carla.WorldSettings(False, False, 0))
    vehicle.destroy()

    write_res(results)
    compare_res()

if __name__ == '__main__':
    main()
```

## Problem
When running the script a few times in a row, I have noticed that the results differ. The difference is not caused by some numeric errors, but it seems like the server does the tick before the processing of the set_velocity instruction is done.

I've looked into the code, and it seems like the set_velocity function does an async call and doesn't wait for the RPC to finish. On the server side we have multiple worker threads running at the same time, which don't seem to care about the order in which the RPC calls came in. 

So here is what I think happens:
* I call set_velocity
* one worker thread A starts processing it
* I call tick (as set_velocity doesn't wait for the result)
* another worker thread B starts processing it
* B finishes before A

If I replace the set_velocity call with a command which I execute in sync (see commented line in example), then I don't encounter the described behavior anymore.

## Questions
1. Is my reasoning described above correct (about why the described behavior happens)?
2. If yes, can we describe this in the documentation, as it might come in handy if someone wants to have reproducible results (for tests or something similar)?

A similar thing happens if I use apply_control, where the docs state 
> Applies a control object on the next tick

which is then wrong as it might not always be applied on the next tick.

## Additional info
**IMPORTANT**: To get the script to run properly, you need reference data. Just run it once, ignore the exception and rename the created results.txt file into reference.txt. Afterwards the comparison will finish properly.

You might also need to run it a few times to see the described behavior. This might be useful, if you are on windows:

```bash
FOR /L %v IN (1,1,50) DO python tick_hangs.py
```

## Environment
OS: Win 10
Python: 3.7
Carla versions: 
- 0.9.7 (locally built from 1f7669e4b308aa8e8f7612dde6666ba6eda14e9f) - tried both the python and C++ API
- 0.9.8 (official package) - tried only the python API
- 0.9.9 (official package) - tried only the python API

germanros1987, 2020-06-26:
@jackbart94 we need your help on this. Could you please take a look?

jackbart94, 2020-06-26:
Hi,

First of all, I've been testing your script following your instructions and I always get ``Results True``. I've tried more than 20 times. I'm on current master (which is a bit after 0.9.9.4) and Ubuntu 16.04 with Python 3.5.2. But I reckon that your issue might be legitimate.

Having said so, one thing I noticed (even if it's commented) is that you use:
``        # client.apply_batch_sync([carla.command.ApplyVelocity(vehicle.id, carla.Vector3D(y=velocity[index]))])``
But apply_batch_sync has by default the False flag, so to make it work in sync mode properly you have to do:
``        # client.apply_batch_sync([carla.command.ApplyVelocity(vehicle.id, carla.Vector3D(y=velocity[index]))], True)``
Can you let me know if this by any chance solves your problem?

Vaan5, 2020-07-08:
Hi,

sorry for the late reply. I was burried in work the last 2 weeks and didn't have time to try it out. Hope it's not too late.

So, I've tried it again today with:

> Carla 0.9.9.4 (downloaded from the github link)
> Python 3.7.7
> Windows 10  (sry I don't have an ubuntu so I can't try it on one)

Here is exactly what i did:
1. Start Carla Server
2. Open console; add the carla egg to the PYTHONPATH variable
3. Run python results.py   // I named the script results.py (i just copied the code from my first post into it - **without any modifications**)
4. Renamed the created results.txt to reference.txt
5. Executed the script 100 times in a loop (FOR /L %v IN (1,1,2) DO python results.py)

I've got Results False 4 times in those 100 runs (in a second try it was 11 out of 100). I don't know, maybe your machine is faster than mine or some linux specifics kick in and that's why you don't get it. I've also attached a txt file with the output of the command from the 5th step: [out.txt](https://github.com/carla-simulator/carla/files/4890095/out.txt)

Regarding your question about client.apply_batch_sync:
1. Just to describe you the workflow how i encountered the problem.:
At first, I didn't use the apply_batch_sync commands at all. I used the setter-like functions (set_location, apply_vehicle_control etc.). I created a few automated tests where i compared the results, and that's how i noticed the differences. Then, I started looking into the Carla code, and saw that the setter functions never wait for a response from the server. The apply_batch_sync method will actually wait for a response. That's how i got to that. After i changed my code to use the apply_batch_sync functions, the non-determinism in the results was gone. The **Problem** part of my original post should describe my way of thinking there in more detail.

2. Regarding the due_tick_cue argument. I still want to be able to call tick explicitly. (one use case for example: 2 clients - one does the ticks, the other controls the car), i.e. I need to have the following sequence in the code:
```
while True:
    apply_batch_sync()
    world.tick()
```
Now if in this sequence I set the argument to True, doesn't that mean that in one loop iteration 2 ticks would happen (one implicitly triggered by apply_batch_sync and the other by world.tick()? That is not something i want/need as i want one snapshot per iteration of the loop.

3. Or did you mean with your comment about the flag (True/False) that apply_batch_sync **can't** be used in synchronous mode without that flag being set to True? If so, could you please elaborate why as I don't really get how that flags fits into the example from the original post.

**Important**: I originally intended the issue to be more like a question (and if it turns out that my thinking is correct, we might adapt the docs, or create a bug issue if it turns out to be one). I didn't get a response on the discord channel, so that's why I created the issue. Please see the **Questions** part of my first post! My main doubt was whether the the setter-like functions were even intended to be used in synchronous mode wrt. determinism.

Now I see that it is marked as a bug. So i am not sure I understand what exactly you r looking at right now.

_Edit_:
Regarding your request/question:
> Having said so, one thing I noticed (even if it's commented) is that you use:
> # client.apply_batch_sync([carla.command.ApplyVelocity(vehicle.id, carla.Vector3D(y=velocity[index]))])
> But [...]

I've tried it out with the flag being set to True, and it behaves exactly as I have described it point 2 (above). 2 ticks would happen in one iteration loop, and that wouldn't work.

closed 2020-09-14 by stale[bot]

Vaan5, 2020-12-01:
Any news regarding this ?