carla-simulator/carla #2815: no installation candidate

spiegelss, 2020-05-04:
Good morning,

I'm trying the quick install and am on the ```sudo apt-get install carla```.  I get the following error message:

```Reading state information... Done
Package carla is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'carla' has no installation candidate
```
This is after adding carla 0.9.8 repository.  Am I doing something wrong?  Thanks!

spiegelss, 2020-05-05:
I've compiled it from source as a solution but I'll leave this open in case anyone else experiences similar issues.

kannidekan, 2020-06-08:
try this command:
`sudo apt-get install carla-simulator`
not:
`sudo apt-get install carla`

tiger-bug, 2020-06-12:
```sudo apt-get install carla-simulator``` worked.  Thank you!

closed 2020-06-12 by tiger-bug

leonpano2006, 2022-02-21:
``┌──(leonpano㉿leonpano-MacBookPro)-[~/Downloads]
└─$ sudo apt install carla
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package carla is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'carla' has no installation candidate
                                                                                                                                                                                                                                                                                                                           
┌──(leonpano㉿leonpano-MacBookPro)-[~/Downloads]
└─$ sudo apt-get install carla-simulator
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package carla-simulator``

revanthsenthil, 2023-04-17:
Any fixes on this? I have the error `E: Unable to locate package carla-simulator` prompted.