carla-simulator/carla #2816: Controlling Pedestrian Traveling Path

KasraMokhtari2112, 2020-05-04:


germanros1987, 2020-06-26:
@KasraMokhtari2112 the pedestrian will go to that point if it's possible. It will try to find a way. If you want to have full control of the path, then you should use carla.WalkerControl.

@bernatx do you have anything to add?

bernatx, 2020-07-02:
Hi,
We have two controllers to be applied to pedestrians:
- **WalkerController**: this is to manage the pedestrian as a vehicle in manual control. You can define the direction and speed to move, and also you can jump. This is used in the **manual_control.py** script.

- **WalkerAIController**: this is to manage the pedestrian to move automatically without user intervention and going to random locations in sequence. It is fine to populate the map. This is used with the **spawn_npc.py** script.

closed 2020-07-30 by germanros1987

chirudeep001, 2023-09-06:
@germanros1987 @bernatx  Can we also use WalkerAIController to trigger the walker when the vehcile reaches certain loctiaon.
Usually, for scenario generation. Does it work??