carla-simulator/carla #2817: How to convert pointcloud from depth image to world frame

Saminhasan5220, 2020-05-04:
I am using carla 0.9.8

and using the following code - 

	#!/usr/bin/env python

	# Copyright (c) 2019 Aptiv
	#
	# This work is licensed under the terms of the MIT license.
	# For a copy, see <https://opensource.org/licenses/MIT>.

	"""
	An example of client-side with basic car controls.

	Controls:

		W			 : throttle
		S			 : brake
		AD			 : steer
		Space		 : hand-brake

		ESC			 : quit
	"""

	# ==============================================================================
	# -- find carla module ---------------------------------------------------------
	# ==============================================================================

	import time
	start = time.time()
	import glob
	import os
	import sys

	try:
		sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
			sys.version_info.major,
			sys.version_info.minor,
			'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
	except IndexError:
		pass

	# ==============================================================================
	# -- imports -------------------------------------------------------------------
	# ==============================================================================

	import carla
	import weakref
	import random
	from math import pi, radians, sin, cos, tan
	import open3d as o3d

	import pygame
	from pygame.locals import K_ESCAPE
	from pygame.locals import K_SPACE
	from pygame.locals import K_a
	from pygame.locals import K_d
	from pygame.locals import K_s
	from pygame.locals import K_w
	from pygame.locals import K_m
	import numpy as np
	from numpy.matlib import repmat

	VIEW_WIDTH = 1920//3
	VIEW_HEIGHT = 1080//3
	VIEW_FOV = 90

	# ==============================================================================
	# -- BasicSynchronousClient ----------------------------------------------------
	# ==============================================================================
[... 333 lines cut ...]
				self.car.destroy()
				self.depth_display.destroy_window()
				pygame.quit()

	# ==============================================================================
	# -- main() --------------------------------------------------------------------
	# ==============================================================================

	def main():
		"""
		Initializes the client-side bounding box demo.
		"""

		try:
			client = BasicSynchronousClient()
			client.game_loop()
		finally:
			print('\n','EXIT')

	if __name__ == '__main__':
		main()

but having problem to transform the local pointcloud in global frame.

How can i construct the transformation matrix?

germanros1987, 2020-06-26:
@DSantosO could you assist with this?

DSantosO, 2020-06-26:
Hello Saminhasan5220,
Thanks for your comment.
We have noticed a problem in the coordinate system in which the point cloud is given. It was not in the correct sensor frame. We have fixed the issue and it is currently under review in a PR so it should be in master very soon. You can check it already in the dsantos/lidar-transform branch. Also, with a several other improvements for the LiDAR sensor, it will be available in the next release.
In that PR, we have also exposed the transformation 4-matrix and its inverse so it is easier to transform from the sensor's frame to the world's one. I think with that fix and that matrix transformation you should be able to achieve your goal.
We will add an script example but I this should be helpful to do that in the mentioned branch:

points = np.frombuffer(image.raw_data, dtype=np.dtype('f4'))
points = np.reshape(points, (int(points.shape[0] / 3), 3))
points = np.append(points, np.ones((points.shape[0], 1)), axis=1)
transform = np.array(sensor.get_transform().get_matrix())
points = np.dot(transform, points.T).T
points = points[:, :-1]

closed 2020-07-02 by marcgpuig

dibenedetto, 2023-10-21:
> Hello Saminhasan5220, Thanks for your comment. We have noticed a problem in the coordinate system in which the point cloud is given. It was not in the correct sensor frame. We have fixed the issue [...]

hello,
i'm trying to use depth buffer values to reconstruct a dense point cloud of some part of the map by looking at that part of the scene from multiple points of view.
however, when i basically use the OP code to go from depth to camera space 3D points, and then use your code from camera to world space, from two similar point of views i obtain the expected point cloud, but they are completely misaligned, as you can se from the snapshot, it's like some hidden transform is applied.

do you have some advice on this issue?

[image]