carla-simulator/carla #2818: PythonAPI Examples not responding

Okeef, 2020-05-05:
Hi Everyone !

I am just starting to use this program. I entered the carla.org website. After i clicked get started and downloaded Python 3.8.3rc1-amd64 , numpy-1.18.4-cp38-cp38-win_amd64.whl , pygame-1.9.6-cp38-cp38-win_amd64.whl , Carla 0.9.9. I installed both modules using pip, i runned the commands. And i entered the CarlaUE4. I wrote on the command line.

> cd PythonAPI/examples

> python3 spawn_npc.py

But it didn't responding there was no change in the CarlaUE4. Command Window didn't say successful or unsuccessful.
[image]

germanros1987, 2020-06-26:
@Okeef could you please provide more information? Is the server running? Does manual_control work for you?

germanros1987, 2020-07-30:
Issue closed due to lack of activity. Feel free to open a new issue if you still consider this is a critical matter.

closed 2020-07-30 by germanros1987

ISUSunny, 2020-08-07:
Hi, guys,

I have met exactly the same issue. I followed all the instructions of building CARLA on Windows. Both "make launch" and "make PythonAPI" seem successful, but there is no response to the command "python3 spawn_npc.py". However, "spawn_npc.py" can be manually started and works well. I am working on Windows 10. Any idea or thought?

[image]

Thanks