carla-simulator/carla #2819: NPC vehicles not moving

Roosstef, 2020-05-05:
Hi all,

since updating to 0.9.9 the Vehicles from spawn_npc.py wont move anymore. (The pedestrians are still moving) When using scenario runner the target vehicles in these scenarios arent moving.
Is there a new way to control npc vehicles?

Greetings
Stefan

jackbart94, 2020-05-05:
Hi,

About spawn_npc, that's weird, can you give us some more information about this? What are you executing exactly? Which OS are you on? Which version of Python? Is anything happening on the terminal? Also in scenario_runner, what are you running exactly?

Roosstef, 2020-05-05:
Hi,

im using ubuntu 18.04 with python2.7. I run a CarlaUE4.sh server on a remote maschine, manual_control.py from examples of the latest master and spawn_npc.py from latest master.

Everything worked fine with the last carla version using same setup and scripts.

I run scenario_runner using the carla_ros_brige carla_ad_demo.launch file. Everything works properly except the target vehicle is not moving.

In both cases the consol is throwing no errors.

Greetings stefan

Roosstef, 2020-05-12:
Anyone has the same problem as me?

faultfactory, 2020-06-09:
Same.
very similar setup. Remote machine, headless, ssh to start the carla server and run the spawn_npc.py script.  Cars do not move. I am running in synchronous mode. 

I did have them moving at one point and the output to ros looked like there are two client ticks which i believe to be incorrect as the ros bridge output was filled with warnings. 
The spawn npc tick would move the cars forward and then the ros bridge client would move the ego forward. 

Through some miracle of button mashing, now i don't have any movement on the npcs. It seemed to be very dependent on order of operations so i'm trying to figure that out as well.

germanros1987, 2020-06-26:
@joel-mb any insights?

joel-mb, 2020-06-30:
Hi @Roosstef @faultfactory , are you still facing this issue? We've done major modifications to the traffic manager since this issue was opened.

faultfactory, 2020-06-30:
Hi @joel-mb. Thank you for the follow up. I will try and squeak in some time to test and validate today for you. 

Were these changes in 0.9.9.4?  There was a single instance of no NPC vehicle movement with this. It's hard to recreate this. Also there were issues with the odometry reported via the ros bridge when the system did run with everything in motion. 

If you want me to test something after that tag, would pulling the carla:latest docker work?

shuhangchen, 2020-07-01:
> Hi @Roosstef @faultfactory , are you still facing this issue? We've done major modifications to the traffic manager since this issue was opened.

Hi @joel-mb , I am facing the same issue. I think it has something to do with the traffic manager? In my case, sometimes the program works well, sometimes the car does not move and the portion of my codes that computes control are not being execuated at all (I have print function inside the 'compute-control' part, but nothing is printed on the terminal).

And which version of carla you were referring to when you said the traffic manager is updated?

Thanks a lot!

joel-mb, 2020-07-02:
Hi!
@faultfactory @shuhangchen I'm trying to set up the same configuration to figure out what is happening. The major updates of the TM were introduced in the release 0.9.9.3 (we introduced traffic manager 2.0).

@faultfactory yes, carla:latest should work.

Roosstef, 2020-07-07:
I no longer have the problem. Thank you!

closed 2020-09-12 by stale[bot]