carla-simulator/carla #2820: Error

wendersonj, 2020-05-05:
Trying to run Carla 0.9.9 in Google Cloud Engine with command

>  DISPLAY= ./CarlaUE4.sh -opengl

and receiving 

> 4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default

and when closing with ctrl+c

> ^CFUnixPlatformMisc::RequestExitWithStatus
FUnixPlatformMisc::RequestExit
Exiting abnormally (error code: 130)

it's a error of missing library or something else ?

wendersonj, 2020-05-05:
Just installed library `alsa-utils` but error persists.

wendersonj, 2020-05-05:
solved with the help of 
[https://medium.com/@etendue2013/how-to-run-ai2-thor-simulation-fast-with-google-cloud-platform-gcp-c9fcde213a4a](url)

> The workaround to prohibit the above message is to create a file /etc/asound.conf with the following content. A detailed explanation may be found in this link.

`pcm.!default {
  type plug
  slave.pcm "null"
}`

closed 2020-05-05 by wendersonj

zhukevin04, 2020-08-07:
> Just installed library `alsa-utils` but error persists.

Hi, i am also using Carla 0.9.9 on GCP, do i need insta lllibrary `alsa-utils` before i follow the medium's method? thks

wendersonj, 2020-08-07:
> > Just installed library `alsa-utils` but error persists.
> 
> Hi, i am also using Carla 0.9.9 on GCP, do i need insta lllibrary `alsa-utils` before i follow the medium's method? thks

I can't say if It will work without install `alsa-utils`, but I did installed and created the file with the content from the previous message. That said, I suggest you create the file (without the `alsa-utils`) , try to run Carla and post for us your results, if you succeed or not.

zhukevin04, 2020-08-07:
> I can't say if It will work without install `alsa-utils`, but I did installed and created the file with the content from the previous message. That said, I suggest you create the file (without the [...]

i add `pcm.!default { type plug slave.pcm "null" }` to `asound.conf` to prohibit the ALSA lib message, but the Xorg Dispaly still some problem, and can't the examples in PythonAPI. Did you success run the examples on GCP?

wendersonj, 2020-08-07:
Yes. After this problem, everything went fine. Right now I'm creating a new project to restart my tests with my code for Carla in GCP. If you installed the nvidia's drivers correctly (check it !), you shouldn't have problems... I suggest that you search for you problem here in the github (it can be something else) and if you don't find it, create a topic. 
cheers!

zhukevin04, 2020-08-09:
> Yes. After this problem, everything went fine. Right now I'm creating a new project to restart my tests with my code for Carla in GCP. If you installed the nvidia's drivers correctly (check it !), [...]

Aha, so you mean you can see the UE4 window? or just run the program, i want to the UE4 window on GPC through Xorg. Thks!

wendersonj, 2020-08-09:
> > Yes. After this problem, everything went fine. Right now I'm creating a new project to restart my tests with my code for Carla in GCP. If you installed the nvidia's drivers correctly (check it !), you shouldn't have problems... I suggest that you search for you problem here in the github (it can be something else) and if you don't find it, create a topic.
> > cheers!
> 
> Aha, so you mean you can see the UE4 window? or just run the program, i want to the UE4 window on GCP through Xorg. Thks!
> 
> When i launch Xorg, there is an error `(EE) Fatal server error (EE) no screens found`
> Then i use P100 with CUDA10.2 and nvidia-smi works.

Hi ! I don't if I understood correctly your question ... BUT:

I can't see any graphic interface, only information from the terminal (text). I think there is a way, using different remote access with, but I never used or have more information). I don't launch Xorg, just run Carla and my scripts. 

Ps: Nvidia's drivers are important to run the server and won't work without them.

zhukevin04, 2020-08-09:
> Hi ! I don't if I understood correctly your question ... BUT:
>
> I can't see any graphic interface, only information from the terminal (text). I think there is a way, using different remote access [...]

Hi, i understand that! Thanks! I try to see the graphic interface but still some errors, i am still trying to solve it, if success i will come back.

qhaas, 2020-11-25:
It is puzzling how this appears in CARLA 0.9.8 when built from source with `make package`, but not in the official CARLA 0.9.8 docker image...

wendersonj, 2020-11-25:
Probably the ubuntu image from docker already was configured and your computer it's not.

Stephanehk, 2020-12-05:
I understand that `adding pcm.!default { type plug slave.pcm "null" }` to `/etc/asound.conf` should solve the problem, but I am a little confused because no such directory exists in my Carla installation. Do I need to create it, and if so where? Thanks!

wendersonj, 2020-12-05:
> I understand that `adding pcm.!default { type plug slave.pcm "null" }` to `/etc/asound.conf` should solve the problem, but I am a little confused because no such directory exists in my Carla installation. Do I need to create it, and if so where? Thanks!

@Stephanehk I don't understand what you mean by doesn't exist in your Carla installation, because the file `asound.conf` is part of the file system `/etc/`. In fact, it's a library. If it doesn't exist, you should install the library `alsa-utils`, as I said previously. For further explanation of why make that change in the file check it out: `https://medium.com/@etendue2013/how-to-run-ai2-thor-simulation-fast-with-google-cloud-platform-gcp-c9fcde213a4a`.

Cheers.

Stephanehk, 2020-12-05:
@wendersonj Is this necessary to start the Carla server? I get the above error messages and my server does not run, but I am not sure if it's a problem with Alsa or if something else is going on.

wendersonj, 2020-12-06:
Nop. You should know if it's a alsa problem if it shows up any message about alsa or related... In case you are no able to start your server, you should look up what it is in your log and search for more information here on issue's github from Carla.

Stephanehk, 2020-12-06:
@wendersonj I ask this because I get the error messages above and then nothing happens, Carla just hangs. Because no other error messages are printed (even though Carla does not terminate), I am unsure if the server is not starting because of Alsa. Do you have any ideas?

wendersonj, 2020-12-06:
You should post a print for your terminal.

Stephanehk, 2020-12-06:
@wendersonj  My terminal displays exactly the same error message you had:

```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
```

wendersonj, 2020-12-06:
> I understand that `adding pcm.!default { type plug slave.pcm "null" }` to `/etc/asound.conf` should solve the problem, but I am a little confused because no such directory exists in my Carla installation. Do I need to create it, and if so where? Thanks!

When you said that you didn't have the folder, did you install `alsa` after ? After installing alsa, you should do the changes on the file .. Have you done it?

> solved with the help of
> [https://medium.com/@etendue2013/how-to-run-ai2-thor-simulation-fast-with-google-cloud-platform-gcp-c9fcde213a4a](url)
>
>
> `pcm.!default { type plug slave.pcm "null" }`

Stephanehk, 2020-12-06:
@wendersonj I was trying to avoid installing Alsa because I do not have root privileges, however, if that is the only way to resolve the issue I will give it a shot. Thanks!

wendersonj, 2020-12-06:
okay ! I don't know any workaround without installing, but try install it ... or if you find any other way, post here ! 

cheers

Stephanehk, 2020-12-07:
@wendersonj  First off I believe I found a better way to do this. I simply created a file entitled .asoundrc in my home directory and copy and pasted `pcm.!default { type plug slave.pcm "null" }` into this file [(more details here)](https://www.alsa-project.org/wiki/Asoundrc). I now however still have the error 
`xcb_connection_has_error() returned true
Failed to create secure directory (/run/user/40445/pulse): No such file or directory`

which seems to have to do with the Pulse audio library. Do you have any ideas or does this warrant a new issue?

wendersonj, 2020-12-07:
@Stephanehk Great ! Thanks for share your workaround. But, this probably it's a new error and I don't have ideas... A quickly search on google, I found this: [https://unix.stackexchange.com/questions/320809/audio-not-working-on-debian-jessie](url). Not sure if will help you. Sorry.

Stephanehk, 2020-12-07:
@wendersonj yeah, the CARLA server just hangs as yours did originally. I still don't understand why because I would not think audio is necessary for it to run.

wendersonj, 2020-12-10:
I think that there was a parameter `-no-sound`, but I don't know if it works...

jewes, 2021-07-29:
> I think that there was a parameter `-no-sound`, but I don't know if it works...

The option should be `-nosound`, and it works.