carla-simulator/carla #2821: Stop lines locations

MohamedDaoud1, 2020-05-06:
Is there a way to get stop lines locations on the map like waypoints?

germanros1987, 2020-06-26:
You can retrieve the stopping volume-boxes using the API. @glopezdiest could you provide a brief code example?

glopezdiest, 2020-06-26:
Hey @MohamedDaoud1. Unfortunately no, there is no way to get the stop lines. The only thing that comes close to it is to get the last waypoint that is part of the road before the intersection starts. Here is an example code of how you'd do it ([traffic_waypoints.zip](https://github.com/carla-simulator/carla/files/4835695/traffic_waypoints.zip))

What this does is
1) Get all the traffic lights and stops
2) Get the trigger volume of the traffic light.
3) Separate it into points to get all the waypoints of the affected lanes
4) Move all the waypoints forward until an intersection is found.
5) Plot the waypoints and store them in a list

This is sometimes good, but depending on the intersection, this might not be accurate. For example, at Town03, the interior part is good, but the exterior fails to mark the stop lines.

[image]

[image]

closed 2020-08-03 by glopezdiest

Khatere1373, 2023-02-14:
> Hey @MohamedDaoud1. Unfortunately no, there is no way to get the stop lines. The only thing that comes close to it is to get the last waypoint that is part of the road before the intersection [...]

Hi, how can I get the location of each intersection on Carla map ?