carla-simulator/carla #2822: Unexpected behavior with zero distortion at semantic segmentation and depth sensors

Sid1057, 2020-05-06:
Hello!

When in synchronous mode I set the distortion to zero for semantic segmentation and camera depth, I get a too strange result:
```
        self.bp.set_attribute('lens_k', str(0.0))
        self.bp.set_attribute('lens_circle_falloff', str(0.0))
```
BGR:
[image]

Depth (BGR):
[image]

Semantic (cityscapes):
[image]

But if I try to set the distortion too small, but not zero:
```
        self.bp.set_attribute('lens_k', str(0.000001))
        self.bp.set_attribute('lens_circle_falloff', str(0.000001))
```
BGR:
[image]
Depth (BGR):
[image]

Semantic (cityscapes):
[image]

As you can see, color camera frames are ok in both cases, but something happens with another sensors.

Ubuntu 18.04; Carla 0.9.9 compiled release

germanros1987, 2020-06-26:
@marcgpuig could you look into this?

marcgpuig, 2020-07-03:
Hi @Sid1057, sorry for the late response.
I've tested it and it works as expected for me, and this
```py
self.bp.set_attribute('lens_k', str(0.0))
self.bp.set_attribute('lens_circle_falloff', str(0.0))
```
provides me with this single color screen in RGB, Sem seg, and depth that you reported (which is the expected output for these values).
If you want to _deactivate_ the effect, please consider setting the values to the default ones that you can find in our [documentation](https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera).
Yet another way to find these default values is using the `carla.ActorAttribute.recommended_values` where you can find a list of recommended values for the sensor, in this case, I set the default values as a recommended values so you will always find them there.
An example, getting the default value:
```py
self.bp.get_attribute('lens_circle_multiplier').recommended_values[0]
```

marcgpuig, 2020-07-03:
Hi @Sid1057, sorry for the late response.
I've tested it and it works as expected for me, and this
```py
self.bp.set_attribute('lens_k', str(0.0))
self.bp.set_attribute('lens_circle_falloff', str(0.0))
```
provides me with this single color screen in RGB, Sem seg, and depth that you reported (which is the expected output for these values).
If you want to _deactivate_ the effect, please consider setting the values to the **actual** default ones that you can find in our [documentation](https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera).
Yet another way to find these default values is using the `carla.ActorAttribute.recommended_values` where you can find a list of recommended values for the sensor, in this case, I set the default values as a recommended values so you will always find them there.
An example, getting the default value:
```py
self.bp.get_attribute('lens_circle_falloff').recommended_values[0]
```

closed 2020-09-12 by stale[bot]