carla-simulator/carla #2824: Carla crashes with Segmentation Fault error with the new Opendrive tool.

aadityasp1, 2020-05-06:
Hi, I'm trying out the Opendrive standalone tool on Carla 0.9.9. Compiled version. (using NVIDIA 1080Ti Gpu, Ubuntu 16.04) 
When I run ./CarlaUE4.sh, server opens up fine, but when I try to connect to it with config.py client under PythonAPI/Util with a the new - x  abc.xodr. The server window crashes. With the following error.

`
Disabling core dumps. 
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=107920 LargeMemoryPoolOffset=239056 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped) `

What could be the reason for crash? I tried with ./CarlaUE4.sh - opengl and  ./CarlaUE4.sh  -opengl3  args also, but it still crashes.

ocean1211, 2020-05-15:
The same problem.
And the Nvidia driver for me 440, OS is 18.04, OpenGL is 4.6
It always raises the error: signal 11 when I spawn the pedestrians.

Here is the detailed report:
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 772] 
VulkanRHI::vkCreateBufferView(GetParent()->GetInstanceHandle(), &ViewInfo, VULKAN_CPU_ALLOCATOR, &View) failed, VkResult=-2
 at /home/jenkins/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:1423 
 with error VK_ERROR_OUT_OF_DEVICE_MEMORY
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=148816 LargeMemoryPoolOffset=279952 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

Is there any solution?

germanros1987, 2020-06-26:
let's dig into this. @bernatx @marcgpuig could you take care of this?

bernatx, 2020-07-02:
Hi,
All seems that the .xodr could be malformed. Can you share that file? Have you tried with the last patch version 0.9.9.4?
Regards

marcgpuig, 2020-07-03:
@aadityasp1 how big is your OpenDRIVE file in MB and/or km^2?

aadityasp1, 2020-07-06:
Hi @bernatx I have tried with the latest 0.9.9.4 patch version as well, still throws the same Segmentation fault error. Attaching two of the xodr's below FYR.

@marcgpuig , I have tried using various sized maps. Below I'm attaching 2 files FYR, one is 5MB, other is 20MB and both throw the same error. Not really sure about km^2 though.

[xodrs.zip](https://github.com/carla-simulator/carla/files/4876711/xodrs.zip)

closed 2020-09-12 by stale[bot]

hongyuli, 2021-01-21:
Same problem here, any updates?

bernatx, 2021-01-25:
Hi, we have been checking the .xodr and they seem to be wrong. 
At some point there are some lanes trying to link to non existing lanes on other roads. That could be a problem by the conversion process used. 
For example, in Bangalore_TEST_map.xodr file:

The road 6994 tries to link its lane -1 to the road 5880 with lane -2. 
But road 5880 has no lane -2. 

In the juntcion 2 it makes the link between them:
`    <junction name="1837595273" id="2">
        ...
        <connection id="4" incomingRoad="5880" connectingRoad="6994" contactPoint="start">
            <laneLink from="-1" to="-2"/>
        </connection>
    </junction>
`
Neither of those roads has a lane -2

Regards

LittleTimeWxk, 2021-12-22:
Same problem here, any updates?

aruckman2, 2022-11-15:
I'm experiencing this on carla 0.9.13 with .xodr file