carla-simulator/carla #2825: make LibCarla fails because of [-Werror,-Wimplicit-float-conversion]

ptigas, 2020-05-06:
Hi,

Building LibCarla (master branch) fails because of strictness of compiler flags. 

```
-- Generating done
-- Build files have been written to: /scratch/pankas/carla/Build/libcarla-client-build.debug
[37/117] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/geom/MeshFactory.cpp.o
FAILED: LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/geom/MeshFactory.cpp.o
/usr/bin/clang++-8   -DBOOST_ASIO_ENABLE_BUFFER_DEBUGGING -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_TEST_CONTENT_FOLDER=\"/scratch/pankas/carla/Build/test-content\" -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libstdcxx-install/include -isystem ../recast-cdce4e-c8-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion   -g -MMD -MT LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/geom/MeshFactory.cpp.o -MF LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/geom/MeshFactory.cpp.o.d -o LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/geom/MeshFactory.cpp.o -c ../../LibCarla/source/carla/geom/MeshFactory.cpp
../../LibCarla/source/carla/geom/MeshFactory.cpp:371:8: error: using integer absolute value function 'abs' when argument is of floating point type [-Werror,-Wabsolute-value]
    if(abs(distance3D) < EPSILON) {
       ^
../../LibCarla/source/carla/geom/MeshFactory.cpp:371:8: note: use function 'std::abs' instead
    if(abs(distance3D) < EPSILON) {
       ^~~
       std::abs
../../LibCarla/source/carla/geom/MeshFactory.cpp:371:12: error: implicit conversion turns floating-point number into integer: 'const float' to 'int' [-Werror,-Wfloat-conversion]
    if(abs(distance3D) < EPSILON) {
       ~~~ ^~~~~~~~~~
2 errors generated.
[38/117] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/nav/Navigation.cpp.o
FAILED: LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/nav/Navigation.cpp.o
/usr/bin/clang++-8   -DBOOST_ASIO_ENABLE_BUFFER_DEBUGGING -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_TEST_CONTENT_FOLDER=\"/scratch/pankas/carla/Build/test-content\" -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libstdcxx-install/include -isystem ../recast-cdce4e-c8-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion   -g -MMD -MT LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/nav/Navigation.cpp.o -MF LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/nav/Navigation.cpp.o.d -o LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/nav/Navigation.cpp.o -c ../../LibCarla/source/carla/nav/Navigation.cpp
../../LibCarla/source/carla/nav/Navigation.cpp:962:74: error: implicit conversion loses floating-point precision: 'double' to 'float' [-Werror,-Wimplicit-float-conversion]
    float shortest_angle = fmod(yaw - _yaw_walkers[id] + 540.0f, 360.0f) - 180.0f;
          ~~~~~~~~~~~~~~   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
../../LibCarla/source/carla/nav/Navigation.cpp:1044:12: error: implicit conversion loses floating-point precision: 'double' to 'float' [-Werror,-Wimplicit-float-conversion]
    return sqrt(agent->vel[0] * agent->vel[0] + agent->vel[1] * agent->vel[1] + agent->vel[2] *
```

Is there something missing from the documentation? Just running `make LibCarla` fails for the right reason.

ptigas, 2020-05-06:
complete log

```
[2/27] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/nav/Navigation.cpp.o
FAILED: LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/nav/Navigation.cpp.o
/usr/bin/clang++-8   -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_TEST_CONTENT_FOLDER=\"/scratch/pankas/carla/Build/test-content\" -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libstdcxx-install/include -isystem ../recast-cdce4e-c8-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion    -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -MMD -MT LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/nav/Navigation.cpp.o -MF LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/nav/Navigation.cpp.o.d -o LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/nav/Navigation.cpp.o -c ../../LibCarla/source/carla/nav/Navigation.cpp
../../LibCarla/source/carla/nav/Navigation.cpp:962:74: error: implicit conversion loses floating-point precision: 'double' to 'float' [-Werror,-Wimplicit-float-conversion]
    float shortest_angle = fmod(yaw - _yaw_walkers[id] + 540.0f, 360.0f) - 180.0f;
          ~~~~~~~~~~~~~~   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
../../LibCarla/source/carla/nav/Navigation.cpp:1044:12: error: implicit conversion loses floating-point precision: 'double' to 'float' [-Werror,-Wimplicit-float-conversion]
    return sqrt(agent->vel[0] * agent->vel[0] + agent->vel[1] * agent->vel[1] + agent->vel[2] *
    ~~~~~~ ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2 errors generated.
[4/27] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/road/MapBuilder.cpp.o
FAILED: LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/road/MapBuilder.cpp.o
/usr/bin/clang++-8   -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_TEST_CONTENT_FOLDER=\"/scratch/pankas/carla/Build/test-content\" -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libstdcxx-install/include -isystem ../recast-cdce4e-c8-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion    -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -MMD -MT LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/road/MapBuilder.cpp.o -MF LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/road/MapBuilder.cpp.o.d -o LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/road/MapBuilder.cpp.o -c ../../LibCarla/source/carla/road/MapBuilder.cpp
../../LibCarla/source/carla/road/MapBuilder.cpp:979:34: error: using integer absolute value function 'abs' when argument is of floating point type [-Werror,-Wabsolute-value]
              static_cast<float>(abs(lane_width))*0.5f;
                                 ^
../../LibCarla/source/carla/road/MapBuilder.cpp:979:34: note: use function 'std::abs' instead
              static_cast<float>(abs(lane_width))*0.5f;
                                 ^~~
                                 std::abs
../../LibCarla/source/carla/road/MapBuilder.cpp:979:38: error: implicit conversion turns floating-point number into integer: 'double' to 'int' [-Werror,-Wfloat-conversion]
              static_cast<float>(abs(lane_width))*0.5f;
                                 ~~~ ^~~~~~~~~~
2 errors generated.
[5/27] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/geom/MeshFactory.cpp.o
FAILED: LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/geom/MeshFactory.cpp.o
/usr/bin/clang++-8   -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_TEST_CONTENT_FOLDER=\"/scratch/pankas/carla/Build/test-content\" -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libs [... cut ...]  carla::geom::Vector3D> > >::value_type' (aka 'std::tuple<unsigned int, boost::shared_ptr<carla::client::Actor>, carla::geom::Vector3D>')
      push_back(value_type&& __x)
      ^
1 error generated.
[9/27] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/trafficmanager/TrafficManager.cpp.o
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:126: recipe for target 'LibCarla.client.release' failed
make: *** [LibCarla.client.release] Error 1
```

germanros1987, 2020-06-26:
@Axel1092 could you help with this?

ForeRunner88, 2020-07-01:
I built Carla-0.9.9 on ubuntu16.04 with `make launch`, but met the same problem above when building client with `make PythonAPI`.

Axel1092, 2020-07-02:
Hi @ForeRunner88,
Your problem is weird, the many of the files compiled for `make launch` are the same for `make PythonAPI`. Could you run `make LibCarla` and send the full log please?

Axel1092, 2020-07-03:
Hi @ptigas,
These compiler errors shouldn't appear, is it possible that you are using mismatched clang versions or the gnu compiler?

ek65, 2020-07-16:
@Axel1092, I am having a similar issue with libcarla but in python setting. Would you mind looking into this error?
I am having issue installing CARLA 0.9.9 PythonAPI in python virtual environment, running on Ubuntu 18.04.4 LTS, Python 3.8, gcc version 7.4.0, clang version 8.0.0.3. The error log is the following:

```
rce/libcarla/libcarla.cpp
running install
running bdist_egg
running egg_info
writing source/carla.egg-info/PKG-INFO
writing dependency_links to source/carla.egg-info/dependency_links.txt
writing top-level names to source/carla.egg-info/top_level.txt
reading manifest file 'source/carla.egg-info/SOURCES.txt'
writing manifest file 'source/carla.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build/lib.linux-x86_64-3.8
creating build/lib.linux-x86_64-3.8/carla
copying source/carla/__init__.py -> build/lib.linux-x86_64-3.8/carla
copying source/carla/command.py -> build/lib.linux-x86_64-3.8/carla
running build_ext
building 'carla.libcarla' extension
creating build/temp.linux-x86_64-3.8
creating build/temp.linux-x86_64-3.8/source
creating build/temp.linux-x86_64-3.8/source/libcarla
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Idependencies/include -I/home/carla_challenge/.cache/pypoetry/virtualenvs/scenic-1WzaWVU2-py3.8/include -I/usr/include/python3.8 -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-3.8/source/libcarla/libcarla.o -isystem dependencies/include/system -fPIC -std=c++14 -Werror -Wall -Wextra -Wpedantic -Wno-self-assign-overloaded -Wdeprecated -Wno-shadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
x86_64-linux-gnu-gcc: error: unrecognized command line option ‘-Wpessimizing-move’
x86_64-linux-gnu-gcc: error: unrecognized command line option ‘-Wduplicate-enum’; did you mean ‘-Wduplicated-cond’?
x86_64-linux-gnu-gcc: error: unrecognized command line option ‘-Wheader-hygiene’
x86_64-linux-gnu-gcc: error: unrecognized command line option ‘-Wfloat-overflow-conversion’; did you mean ‘-Wfloat-conversion’?
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

dHonerkamp, 2020-08-03:
Did you find a solution to this? I'm running into the same problem

Axel1092, 2020-08-20:
@ek65 @dHonerkampIt seems your build system is using gcc when it should be using clang did you run the 
```sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-8/bin/clang++ 180 &&
sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180```
commands as listed in the docs (https://carla.readthedocs.io/en/latest/build_linux/)?

closed 2020-08-26 by Axel1092

jinfagang, 2021-07-19:
@Axel1092 I installed clang and runed `sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180` still get same rror:

```
x86_64-linux-gnu-gcc: error: unrecognized command line option ‘-Wpessimizing-move’
x86_64-linux-gnu-gcc: error: unrecognized command line option ‘-Wduplicate-enum’; did you mean ‘-Wduplicated-cond’?
x86_64-linux-gnu-gcc: error: unrecognized command line option ‘-Wheader-hygiene’
x86_64-linux-gnu-gcc: error: unrecognized command line option ‘-Wfloat-overflow-conversion’; did you mean ‘-Wfloat-conversion’?

```