carla-simulator/carla #2826: Carla 0.9.9.1 debian package crashes on Kubuntu 20.04

derFrickler, 2020-05-07:
When trying to run the latest carla-simulator 0.9.9.1 on Kubuntu 20.04 it crashes with the following error. Installation of the debian package via apt works fine. Running the 0.9.8
 debian package starts up without problems.

`./CarlaUE4.sh 
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
Failed to find symbol file, expected location:
"/opt/carla-simulator/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.sym"
LowLevelFatalError [File:Unknown] [Line: 279] 
Out of Device Memory, Requested=24000.00Kb MemTypeIndex=7
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=145408 LargeMemoryPoolOffset=276544 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
`

derFrickler, 2020-05-07:
Works with the new new GTX 2080TI  Card.

closed 2020-05-07 by derFrickler

qhaas, 2021-03-21:
`Out of Device Memory`
The (which fails) Quadro P400 has 2Gb of memory, the (which works) GTX 2080TI has 11Gb of memory.  CARLA 0.9.9 uses UE 4.24, 0.9.8 uses 4.22.  (speculation) Suspect with the newer engine, more memory was used.  If other users finding this issue experience this, try using `-opengl` instead of vulkan, since [it uses less GPU memory](https://carla.readthedocs.io/en/latest/adv_rendering_options/#vulkan-vs-opengl).