carla-simulator/carla #2829: Visualize Lane Markings in OpenDrive Standalone Mode

EliteTUM, 2020-05-08:
Currently, with version 0.9.9, the OpenDrive standalone mode does not visualize lane markings (e.g. see following image from the [release notes](https://carla.org/2020/04/22/release-0.9.9/) ).

Can they be added? This increases the number of usecases CARLA could be used for. Also it eases transitioning from any other simulator to CARLA, as many other tools already have support of OpenDRIVE. So transferring assets from any other simulator to CARLA would get easier.

[image]

germanros1987, 2020-06-26:
The lane markings are there, but not rendered. We will add them in future versions of the tool.

marcgpuig, 2020-07-01:
Yes, this is in our roadmap. the main difficulties here are how to generate these LaneMarkings as separate meshes of the main road so they can be also added into the semantic segmentation rendering. Since we also want to change how the semantic segmentation works, we must wait until it's completely done so _maybe_ the separation of these meshes will not be necessary.

Kin-Zhang, 2022-03-19:
Is there any update?

soup1997, 2023-03-14:
Is there any progress on this? please let me know.

closed 2024-05-10 by MarcelPiNacy-CVC

rebolomo, 2024-08-09:
Is there any update? please let me know.