carla-simulator/carla #2831: RADAR sensor info

ujjawal-kumar, 2020-05-09:
Hi,
Can some one help me in studying more about RADAR sensor in below aspects. 

(1) it is given that the new model of RADAR sensor added  (https://carla.org/2019/12/11/release-0.9.7/)  is based on Ray tracing. Is it only EM wave reflection based modeling and detection shown in CARLA or it has any association with actual RADAR system level parameters (as asked in below question).

(2) I would like to know what RADAR algorithmic parameters are taken here and how to modify. 

(a) RADAR waveform type (FMCW or which type) waveform parameters such as frequency , bandwidth, chirp rate
(b) Power of transmission
(c)  Detection algorithm parameters such as CFAR threshold etc

(3) I am interested in interference modeling among multiple RADARs.Is there a way to do that. 

(4) I am interested in defining the type of material infront of the RADAR (such as rock, RFC material, metal etc)
how to do that.

(5) How to control the FOV of the radar in azimuth and elevation.
FOr deeper radar algorithm performance study , i want to keep only few target points and do analysis.

germanros1987, 2020-06-26:
@ujjawal-kumar your initial assumptions are incorrect.

The RADAR model that is currently available in CARLA is a placeholder. It is not based on raytracing...it is based on raycasting. 
This RADAR is just casting rays to objects and computing distance and velocity using a simple model.

No FMCW, CFAR threshold, etc. IT IS NOT INTENDED TO BE A PHYSICALLY ACCURATE RADAR.

But don't worry! We are actually working on a physically accurate RADAR as part of our new project: the External Sensor Interface. However, this is still in works. It will take a while. So, please be patient.

closed 2020-06-26 by germanros1987

tharindurmt, 2020-10-14:
@germanros1987 I was wondering whether there is an ETA on the new external senor interface. 
Thanks.

jparismorgan, 2025-01-30:
@germanros1987 curious, do you have a branch / any code you could share on any progress that was made towards a physically accurate radar? Curious about how far we would be away from getting it into carla. Thanks!