carla-simulator/carla #2834: Deployment rotation does not match with desired rotation

adrigrillo, 2020-05-10:
Hello,

I am currently testing the use of testing of static cameras to record some traffic scenes in CARLA 0.99, both in Linux and Windows. 
In concrete, I have a wide range of combinations of positions and rotations.
However, when I set the camera with a pitch of -90 degrees (bird's eye point of view), I am facing the problem that the yaw is combined with the roll once the camera is deployed so I do not get the desired values for both components.

An reproducible example is the following:

```python
# connect to the server and world before this
spectator: carla.Actor = world.get_actors().filter('spectator')[0]
spectator.set_transform(
    carla.Transform(
        carla.Location(0., 0., 50.),
        carla.Rotation(-90., 90., 0.)
    ))
# running loop 
try:
    while True:
        print(f'The location of the spectator is {spectator.get_transform()}.')
        world.wait_for_tick()
except KeyboardInterrupt:
    logger.info('Exit')
```

In my case the result of the print statement is:

`The location of the spectator is Transform(Location(x=0.000000, y=0.000000, z=50.000000), Rotation(pitch=-88.992432, yaw=89.993004, roll=-179.999710))`

So, when I retrieve the yaw, it is 90 degrees and, the roll, -180.
However, I was expecting -90 degrees for the yaw and 0 degrees for the roll.
This case is handling the spectator camera, nonetheless, the same is achieved using cameras and the deploy command.

I know that the result in the image is the same and it is correct, as I have tested it. 
Yet, I also perform operations with respect the yaw of the camera and the objects, being those incorrect.

germanros1987, 2020-06-26:
@Axel1092 could you provide some guidance?

Axel1092, 2020-06-26:
Hi @adrigrillo this behavior is due to the way the Unreal Engine handles rotations. The spectator is a special camera as it is restricted to a range of yaw and pitch values and may have unwanted effects to the final rotation.
I suggest not using the spectator and instead spawn a Camera to do the recording as it will not be subject to such limitations.

adrigrillo, 2020-06-27:
Hi @Axel1092, the same problem is produced when spanning cameras. The solution I adopted was to use -89 degrees for the pitch. If it is an Unreal Engine I guess you could do little to change this, so the issue can be closed.

germanros1987, 2020-06-28:
Is there any way to go around this limitation @Axel1092 ?

Axel1092, 2020-07-02:
After some experimentation it seems that around 90 degrees Unreal will change the values to keep them within certain ranges. I guess the only thing you can do is to try to keep track of the rotations yourself.