carla-simulator/carla #2835: ERROR: Spawn failed because of collision at spawn position

dustyujanin, 2020-05-10:
Hello.
I install 0.9.9 version
When trying to run python3 spawn_npc.py

Error:
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
spawned 80 vehicles and 33 walkers, press Ctrl+C to exit.

jackbart94, 2020-05-11:
Hi,

This message is just a warning that either some vehicles or some walkers (in this case it's probably walkers) haven't spawned because the random location is too near to one of another walker.

So basically, nothing to worry about.

closed 2020-05-11 by jackbart94

adrigrillo, 2020-05-11:
How many vehicles are you trying to spawn at once? For me it seems, that you are trying to spawn more vehicles that spawn points are in the map.