carla-simulator/carla #2840: Running Carla 0.9.6 on VM

Deio828, 2020-05-11:
Hello, 
My local machine can't run carla 0.9.6 properly, the cache size is not compatible to run carla graphics with multiple clients. 
So I'm intending to run Carla on some VM provided by a local company in my country, but they asked me first, what specs should the VM have to run Carla.
Could you please me in order to select the best VM.

faruklinux, 2020-05-22:
Apparently some VMs struggle to support the OpenGL 3 requirements needed to run Unreal Engine, whereas VMs only provide support up to OpenGL 2. I was stuck with an unable to detect Vulkan drivers during my attempts. However, the windows build currently has its own troubles with executing PythonAPI command.

germanros1987, 2020-06-26:
@Deio828 why don't you try AWS? Just use one of our dockers in AWS. You should be good to go.

germanros1987, 2020-07-30:
Issue closed due to lack of activity. Feel free to open a new issue if you still consider this is a critical matter.

closed 2020-07-30 by germanros1987