carla-simulator/carla #2841: Carla cv2.imshow() not working : white screen

dustyujanin, 2020-05-11:
Hello.
I'm trying to run this code: https://pastebin.com/wFM0z1vj
Carla 0.9.9
But cv2.imshow now working. I see only white screen
[image]

pirate-lofy, 2020-05-12:
It might be irrelevant to Calra, but can u try 
    ```
    cv2.imshow("", i3)
    if cv2.waitKey(1)&0xFF==27:
        pass
```

dustyujanin, 2020-05-12:
Still not working:(
[image]
[image]

Hamptonjc, 2020-06-09:
I'm having this issue as well!

wendersonj, 2020-06-09:
I don't have the code right now, but using some other code I had troubles with threads ! Even using a thread worker to do a secondary job, I can't display using either OpenCV or Matplotlib. I'm currently using PyGame, that comes with the basics scripts, but it's more difficult to use.

GRFreire, 2020-06-16:
same here, did you found something?

germanros1987, 2020-06-26:
This is not a CARLA issue. This refers to OpenCV usage.

closed 2020-06-26 by germanros1987

akssieg, 2020-06-27:
I am following the similar code from `https://pythonprogramming.net/control-camera-sensor-self-driving-autonomous-cars-carla-python/?completed=/introduction-self-driving-autonomous-cars-carla-python/`, opencv imshow shows a blank screen and If I try to do imwrite then the files are empty. Has anybody found solution for this problem?

germanros1987, 2020-06-28:
Please, use forum.carla.org for this kind of questions, which are not "purely" about CARLA.