carla-simulator/carla #2842: Lidar attached to vehicle with SpringArm results in limited fov

RyanHopkins7, 2020-05-11:
Hello,

I was following the [lidar raycast sensor guide](https://carla.readthedocs.io/en/latest/tuto_G_retrieve_data/#lidar-raycast-sensor) which suggests using the SpringArm attachment. Using SpringArm results in a non-rotating FOV limited to sight of the front-left and back-right sides of the vehicle as shown below:
[image]
This is what the lidar output of the same scene using the Rigid attachment looks like:
[image]
If I understand SpringArm correctly, it should cause the lidar sensor to avoid entering walls/obstructions and lag slightly behind the vehicle it's attached to rather than arbitrarily limiting the FOV like this. Is this behavior intentional?

germanros1987, 2020-06-26:
@RyanHopkins7 indeed, the springarm provides a very specific pose, which will limit the FOV. I guess, this is not the right tool for your needs. Just use the rigid one.

RyanHopkins7, 2020-06-26:
@germanros1987 All I'm wondering is if this behavior is intentional. If it is, I'll close, but it did take me a little while to figure out the cause of this issue considering that the tutorial I mentioned used SpringArm without limiting the FOV. A note about this in the documentation might be helpful so people know what causes this.

germanros1987, 2020-06-26:
Hold on a second, I may have misunderstood the first screenshot. Is the first image supposed to be taken from a bird-eye-view? I thought the points on the left and right side of the pointcloud were from a wall. Could you further explain the setup, please?

RyanHopkins7, 2020-06-27:
Yes, they were both taken from a bird's eye view using the same settings other than the attachment type. They were also taken in the same environment. The obstructions on the left and right in the first photo are not caused by walls.

germanros1987, 2020-06-27:
@RyanHopkins7 please send over the sensor configuration/setup you have used for this so that we can reproduce it.

@DSantosO once we get the sensor configuration, could you try to reproduce?

RyanHopkins7, 2020-06-27:
This is the sensor configuration I was using:

lidar_cam = None
lidar_bp = world.get_blueprint_library().find('sensor.lidar.ray_cast')
lidar_bp.set_attribute('channels',str(32))
lidar_bp.set_attribute('points_per_second',str(300000))
lidar_bp.set_attribute('rotation_frequency',str(120))
lidar_bp.set_attribute('range',str(20))
lidar_bp.set_attribute('lower_fov',str(-40))

lidar_location = carla.Location(0,0,2)
lidar_rotation = carla.Rotation(0,0,0)
lidar_transform = carla.Transform(lidar_location,lidar_rotation)

lidar_sen = world.spawn_actor(lidar_bp,lidar_transform,attach_to=vehicle2,attachment_type=carla.AttachmentType.SpringArm)
lidar_sen.listen(lambda point_cloud: point_cloud.save_to_disk('sensor_output/lidar/f%.6d.ply' % point_cloud.frame))

DSantosO, 2020-07-01:
@RyanHopkins7 Thank you for the information. I am able to reproduce your result and yes, it seems to be something weird with the output when SpringArm is the attachment type. I will look into it.

DSantosO, 2020-07-06:
Hello, 
Sorry for the late replay, we have looked into the specific of the case and there is couple of things to comment.
First, you need to take in account that the SpringArm is thought to have a better camera visualization so it should not be used in sensors like LiDAR. In this sense, please use always the Rigid mode.
Anyway, the SpringArm is doing something weird in your specific case because of the zero translation in the 'x' axis. This is a problem because this SprintArm is thought to see the car in 'third person' not directly in top of the car. We will add some safeguard about this problem soon.

Thank you very much for reporting the issue, I hope this information is helpful.

RyanHopkins7, 2020-07-06:
@DSantosO Thanks for getting back to me and I'm glad to help. If it's recommended not to use SpringArm for LiDAR, perhaps [the documentation should be changed to demonstrate using the Rigid mode](https://carla.readthedocs.io/en/latest/tuto_G_retrieve_data/#lidar-raycast-sensor).

DSantosO, 2020-07-07:
Hello @RyanHopkins7
Yes, that is mistake. Sorry about it. We will change it.
Thanks again!

PR #3043 (2020-07-15): sergi-e/dif-rigid-spring
#### Description
- Added clarification between Rigid and SpringArm attachments. 
- Fixed instances of declarations in docs. 

Fixes #2842

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04.4
  * **Python version(s):** 3.6.9 and 2.7.17
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks
- LIDAR docs need to be updated.

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3043)
closed 2020-07-09 by marcgpuig