carla-simulator/carla #2843: side road intersection in carla 0.9.5

pirate-lofy, 2020-05-12:
Thanks for Carla team for the great updates in Carla 0.9.x 
in Carla 0.8.x there were some useful features like the offroad_intersection measurement which I can't find in Carla 0.9.5.
Does anyone have any idea how to get this measurement in Carla 0.9.5?

pirate-lofy, 2020-05-29:
Does anybody has any idea how to solve this, please?

germanros1987, 2020-06-26:
Please, do not get too attached to old versions. Why not moving to 0.9.9? 

@glopezdiest could you point @pirate-lofy how to detect the off-road situation?

glopezdiest, 2020-06-26:
Hey @pirate-lofy. According to the docs: 

> These values are computed intersecting the bounding box of the vehicle (as a 2D rectangle) against the map image of the city

I don't think that measurement is available anymore so you'd have to calculate it yourself. You'd need some serious geometry to get the line that intersects the bounding box, and some calculus to get the percentage off-road. There might a library that does this very efficiently but I can't think of one. 

An approximation would be to get the eight points of the bounding box (vertices and middle points) and get how many of the 8 are offroad. To know if you are off_road you can either:
1) get_waypoint(self, location, project_to_road=False), which will return None if the location is outside driving lane
2) get_waypoint(self, location, project_to_road=False, lane_type=carla.LaneType.Any), which will return a waypoint, and if its lane_type is different than *Driving*, the location is out.

pirate-lofy, 2020-06-26:
@glopezdiest  that was so helpful thanks a lot it worked

closed 2020-06-26 by pirate-lofy