carla-simulator/carla #2847: Tuning vehicle dynamics parameters of a new vehicle generated in Unreal Engine

daydin41, 2020-05-12:
Hi, 

We are creating our custom truck vehicle model in UE4 and importing it to CARLA. The tutorial https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/ specifies steps for CARLA import as well as some parameters for vehicle dynamics from blueprint. Are we limited to the vehicle dynamics parameters given in this tutorial? Can furher tuning be done in blueprint?

The built in vehicles seems to use UE4 vehicle's physics (based on PhysX vehicles). If the parameters in PhysX plugin is tuned in UE4 (using the guide: https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/Vehicles.html#tuning-guide) can these parameters be imported to CARLA?

Thanks,

germanros1987, 2020-06-26:
@bernatx could you follow up on this?

bernatx, 2020-07-02:
Hi, because we are starting to integrate vehicle dynamics into CARLA we will check how many of those parameters we can expose to CARLA. Some of those properties are related to art resources (like the wheels size), so it seems to be a complex addition. But we have the focus on it.

yyab, 2020-07-23:
> Hi, because we are starting to integrate vehicle dynamics into CARLA we will check how many of those parameters we can expose to CARLA. Some of those properties are related to art resources (like the wheels size), so it seems to be a complex addition. But we have the focus on it.

Hi! Could you give some information on the integrated vehicle dynamics in Carla? Will that appear in the next Carla release? Are there going to a Pacejka tire model rather than the default tire model in the PhsicX? Thank you!

bernatx, 2020-07-23:
We are working on Carsim integration, but it is not ready for the next release.

yyab, 2020-07-23:
> We are working on Carsim integration, but it is not ready for the next release.

Thank you for your reply! But since Carsim is a licensed product, to use Carsim in Carla we will have to by Carsim license, right? Are there any open-source way to implement more realistic tire model in Carla?

closed 2020-10-04 by stale[bot]

jimaldon, 2022-01-17:
@yyab What would be an open-source alternative to a physics engine for simulating vehicle dynamics like CarSim?